OneOfEleven/uv-k5-firmware-custom #212: Slow memmove() used

f1rmb, 2023-10-24:
Hi,

I was just looking at the code, and I came across this:

https://github.com/OneOfEleven/uv-k5-firmware-custom/blob/024c4bb79de6caba88d65c748b670ad80e3b270c/ui/main.c#L531C9-L531C9

Checking the source and destination buffers, they don't overlap, so memcpy() would be better as memmove() use a third buffer to copy the data from src to dest.

Do you want me to make a PR ?

Cheers.
...
Daniel

OneOfEleven, 2023-10-24:
You'll find I used memmove elsewhere too (or was doing), I was going to use memcpy but simply decided to keep them all as memmove so as not to include another system function (memcpy) so as to save a teeny bit of flash. Probably makes no difference though.

f1rmb, 2023-10-24:
Okay,

   but memcpy() is already used multiple time in the FW, and is faster in the end.
If you want, I can make a pass of small change/optimization (BTW, I'm one of the OpenGD77 dev team).

OneOfEleven, 2023-10-24:
Yes for sure, change the moves to cpy's and upload :)

f1rmb, 2023-10-24:
I can close this none issue now.
Thanks.

closed 2023-10-24 by f1rmb