OneOfEleven/uv-k5-firmware-custom #213: save a few bytes - remove AM Rx fix switch from compile options

pawol, 2023-10-24:
Your AM Rx fix works perfectly so I do not see any sense to maintain option to disable your fix whils compilation.
save a few bytes, make your fix permanent and remove AM Rx fix switch from compile options

OneOfEleven, 2023-10-24:
I have been tempted to do that, but I still get messages (from elsewhere) that say the fix doesn't work, so (up until now anyway) I've left the menu option in to give them the chance to battle with their own internal struggle over it.

The fix does have a limit though when it comes to signal strength, once the fix/code has reached the limit of the RF front end gain adjustment range it can't do anymore to help, at which point the I.F/AM demodulator goes into saturation/overload (distorted audio).

pawol, 2023-10-24:
unbelievable that fox does not works. There is a lot of peple do not having RF technic basics.
I find it works great, especially when I live 30km away from airport and I can heat planes from extremally close and far distances withount any problems. 
There is nothing worse than original AM. So, remove it and forget about it. Everything is better than that

wrcrooks, 2023-10-24:
The AM fix works great for me as well and I think it's fantastic work. Majority rules?