OneOfEleven/uv-k5-firmware-custom #214: Frequency mode scan

krotoniec, 2023-10-24:
I could be missing something here but with the recent versions I can't use scan function in VFO mode anymore. When I hold * key in VFO mode, it then switches to MR mode and runs scan there. How to scan frequencies in VFO mode?

OneOfEleven, 2023-10-24:
oops !

No you're not missing anything (that I know of). It seems to switch to channel mode when I try and start it scanning in frequency mode .. my fault. I was fixing a bug with it yesterday, I've broke it in the process of fixing the other problem.

krotoniec, 2023-10-24:
No worries, I'm just doing my small regression testing not even realizing ;)
Thanks for your work!

OneOfEleven, 2023-10-24:
Should be fixed now. Thank you for letting me know.

closed 2023-10-24 by OneOfEleven