OneOfEleven/uv-k5-firmware-custom #218: Investigate AGC, REG_7E

bricky149, 2023-10-24:
I started looking into the AGC regs as the register table suggests there are tunable params for Rx (IF) and Tx (MIC). As I can see #217 touches code I've made changes to, it may cause a merge conflict.

I'm currently somewhere that has no AM traffic so my changes are untested. Should I raise a PR anyway?

OneOfEleven, 2023-10-24:
> I started looking into the AGC regs as the register table suggests there are tunable params for Rx (IF) and Tx (MIC). As I can see #217 touches code I've made changes to, it may cause a merge [...]

No never do any PR that is not fully tested.

bricky149, 2023-10-24:
> Hi, The AGC code _if_ blocks are just swapped in my PR, and the uint8_t Value argument became a boolean, as the function wasn't doing much with Value except checking for 0 or 1.

The current code keeps AGC disabled in both cases. It matched the DisableAGC code from RT-890's OEFW.

> No never do any PR that is not fully tested.

Understood. The worry is it will go stale and cause problems later on. Should I instead hide it behind a compile flag so it doesn't unintentionally change current behaviour?

OneOfEleven, 2023-10-25:
gggrrr, google have just decided that github mail is all spam and so has been blocking it all for a few hours now :( .. just going through it all, bare with me.

I don't think google approve if github :(

OneOfEleven, 2023-10-25:
Just tried your fix, seems the AM fix is still needed :(

Keep trying though, maybe you'll find the correct register configuration to fix the complete lack of AM agc .. be nice if you find the solution :)

bricky149, 2023-10-25:
> Just tried your fix, seems the AM fix is still needed :(

The potential 'fix' is commented out due to my inability to test it at this time.

My last PR only brings parity with fagci's FW since we weren't making use of AGC at all. I can't play with registers again until I'm surrounded by air traffic again.

OneOfEleven, 2023-10-25:
I noticed that just after I sent the message, my fault

just tryng now

bricky149, 2023-10-25:
> I noticed that just after I sent the message, my fault
> 
> just tryng now

Be aware that enabling it will need a higher squelch setting due to it trying to boost weak signals. I'd suggest disabling the option that makes squelch more sensitive since it affects noise thresholds as well.

OneOfEleven, 2023-10-25:
hmmmm, interesting result using the signal generator !

Just going to leave the radio scanning to find some AM transmissions - can be rare for me too here in shropshire.

bricky149, 2023-10-25:
> hmmmm, interesting result using the signal generator !
> 
> Just going to leave the radio scanning to find some AM transmissions - can be rare for me too here in shropshire.

I had to reduce the 'fix index' from max to min as my radio was finding unintelligible noise in freqs that should actually have activity. That was before I disabled the squelch sensitivity compile option. Hopefully you find something that makes it worth tweaking.

OneOfEleven, 2023-10-25:
I have two UV-K5(8)'s, one seems good on AM and the other has the distorted AM back, same firmware, no AM fix code compiled in, using the AGC enable.

Interesting. Don't know why one works and the other doesn't, wonder if the BK4819 chips are different versions.

bricky149, 2023-10-25:
> I have two UV-K5(8)'s, one seems good on AM and the other has the distorted AM back, same firmware, no AM fix code compiled in, using the AGC enable.
> 
> Interesting. Don't know why one works and the other doesn't, wonder if the BK4819 chips are different versions.

I was looking at BiliBili recently and I suspect there was an audio change to units made after July. I have a K5(8) from June and it's horrible, I have a second one coming that's with the courier.

My stock 5R+ wasn't distorted when I last tried. My curiosity wants to know if it's more than a case acoustics change.

bricky149, 2023-10-25:
I've just noticed the REG_7E values are swapped. '1u << 15' should be in EnableAGC, '0u << 15' in DisableAGC (not called for now). I had them commented out.

bricky149, 2023-10-25:
I'm going to document my results here, now that I know I can hear air traffic.

The filter appears to affect it most.
000 = clicking, voices inaudible
...
100 = whistling, voices understood

bricky149, 2023-10-25:
**TL;DR**

BK4819_WriteRegister(0x7E,
	(1u << 15) |      // 0  AGC fix mode
	(2u << 12) |      // 3  AGC fix index
	(5u <<  3) |      // 5  DC Filter band width for Tx (MIC In)
	(3u <<  0));      // 6  DC Filter band width for Rx (I.F In)

Tx filter bandwidth can be adjusted at a later date, it's outside the scope of what I aimed to achieve.

OneOfEleven, 2023-10-25:
Wonder where you found the filter cutoff vs value details ? .. I've not found those details anywhere I don't think :(

Not sure what you mean "AM shouldn't use oscillation" ?

bricky149, 2023-10-25:
> Wonder where you found the filter cutoff vs value details ? .. I've not found those details anywhere I don't think :(

I inferred the filter values from REG_2B. I can only assume the same layout, based on their defaults.

> Not sure what you mean "AM shouldn't use oscillation" ?

I'm learning on the job. I came across [this](https://www.vintage-radio.com/repair-restore-information/valve_if-rf-stages.html):

"On sets having the AM bands in addition to FM, there will be two IF transformers between each stage of IF amplification. [...] The oscillator section of the AM mixer-oscillator will be disabled, and the mixer section used as the first FM IF amplifier."

OneOfEleven, 2023-10-25:
oh I see.

I think their may be some misunderstanding you have about what they are referring to through the super-het chain.

All RF mixers need a local oscillator, the mixer mixes the RF with the local osc to produce an output at a difference freq. without an osc nothing works.

This is all part of very basic RF theory you learn during college etc.

OneOfEleven, 2023-10-25:
Yes definitely, we ONLY learn by playing, trial and error etc.

OneOfEleven, 2023-10-25:
I've not long realised I've had one of the LNA registers set wrong, have fixed that now.

bricky149, 2023-10-25:
> I've not long realised I've had one of the LNA registers set wrong, have fixed that now.

I was playing with it before I noticed your commit. I told someone already I should stop but I can't help myself.

OneOfEleven, 2023-10-25:
Keep playing !

OneOfEleven, 2023-10-25:
I get things wrong all the time, ask anyone lol  all engineers/coders do.

DualTachyon, 2023-10-25:
Not wrong, just differences of opinion.

bricky149, 2023-10-25:
Seeing as I now properly understand how this register works, I think this issue should be closed. I'd be interested to know how Tx is affected between AGC being on and off, but that's for another day. I appreciate the time and effort spent in looking at this, and the resultant changes.

Thank you.

closed 2023-10-25 by bricky149

OneOfEleven, 2023-10-26:
Don't give up !

The BK chip is a mystery to us, we don't know how to properly configure it as we don't have the details on it's registers, wish we did.

I was battling with the gain/AGC registers for ages before I too gave up and decided to write my own code to overcoming our inability to configure the AGC inside the chip.

It may be that the firmware inside the BK chip has yet to be fixed/updated to deal with things like AM, but there's nothing we can do about that, but it might explain why my two radios react differently to same register settings.

bricky149, 2023-10-26:
I wasn't planning to give up, just a deflating balloon moment finding out it didn't get the results I was looking for. Since other reg values they reference now differ between 'disabled' and 'enabled' state (they were the same before I touched it), maybe they could be useful outside of airband. I think the only main difference I found with fagci's code was PGA gain. Looking at bk4819.c, lines 316 and 355, it's minor but would affect Rx.

Unrelated, my second K5(8) arrived today and it differs to my first. The tinny sound is gone and the aluminium backing plate is now magnesium. I'll have to back up the EEPROM and see what else is different.

bricky149, 2023-10-26:
REG_62 is yielding interesting results. I have two radios set up, each with different AGC on/off settings. I'm getting 20dB(!) difference between them, the one with AGC appears to be attenuating the signal.

I hope to carry this on tomorrow when there's air traffic about. There may still be hope for this rabbit hole.