OneOfEleven/uv-k5-firmware-custom #221: No more firmware.packed.bin?

ct1esj, 2023-10-24:
I, no more firmware.packed.bin so i can download and test?
I'm very basic, ok. But i`m happy :-)

wrcrooks, 2023-10-24:
I believe that OneOfEleven removed the .bin files from the repo back in https://github.com/OneOfEleven/uv-k5-firmware-custom/commit/ff010b9d970adcf935defbbdc9b9118f2ac30534

You'll likely have to compile your own firmware.

Mason10198, 2023-10-25:
You are probably compiling this on Windows and you have the Microsoft Store version of Python installed, causing this error:
```
SyntaxError: Non-UTF-8 code starting with '\x90' in file C:\msys64\mingw64\bin\python.exe on line 1, but no encoding declared;
```
In this case, a quick fix to change the commented lines in your `Makefile` to match this:

```
#	-python fw-pack.py $<.bin $(GIT_HASH) $<.packed.bin
	-python3 fw-pack.py $<.bin $(GIT_HASH) $<.packed.bin
#	-$(PYTHON) fw-pack.py $<.bin $(GIT_HASH) $<.packed.bin
```

OneOfEleven, 2023-10-25:
or no python installed at all.

The compiler creates the firmware.bin file, once that's done, a little bit of python code is run at the end to convert that compiled file into firmware.packed.bin which you can use in the QS windows firmware uploader.

zhaoparrot, 2023-10-25:
FYI I also got some error here 
```
PYTHON = C:\Users\zhao\AppData\Local\Programs\Python\Python311-32\python.exe C:\Users\zhao\AppData\Local\Microsoft\WindowsApps\python.exe
arm-none-eabi-objcopy -O binary firmware firmware.bin
C:\Users\zhao\AppData\Local\Programs\Python\Python311-32\python.exe C:\Users\zhao\AppData\Local\Microsoft\WindowsApps\python.exe fw-pack.py firmware.bin 91b49e9 firmware.packed.bin
C:\Users\zhao\AppData\Local\Programs\Python\Python311-32\python.exe: can't open file 'C:\\Users\\zhao\\AppData\\Local\\Microsoft\\WindowsApps\\python.exe': [Errno 22] Invalid argument
make: [all] 错误 2 (忽略)
```
it will work if i did this 
```
	-python fw-pack.py $<.bin $(GIT_HASH) $<.packed.bin
#	-python3 fw-pack.py $<.bin $(GIT_HASH) $<.packed.bin
#	-$(PYTHON) fw-pack.py $<.bin $(GIT_HASH) $<.packed.bin
```
seems I have two python installed in my laptop...

OneOfEleven, 2023-10-25:
I've reverted that change that was done a while back, seems to cause too many problems for people.

closed 2023-10-25 by ct1esj