OneOfEleven/uv-k5-firmware-custom #227: latest commit cannot compile freq calibration option

pawol, 2023-10-25:
UART -DENABLE_SMALL_BOLD -DENABLE_MUTE_RADIO_FOR_VOICE -DENABLE_VOX -DENABLE_REDUCE_LOW_MID_TX_POWER -DENABLE_TX1750 -DENABLE_MDC1200 -DENABLE_RESET_AES_KEY -DENABLE_KEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_F_CAL_MENU -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_AM_FIX -DENABLE_FASTER_CHANNEL_SCAN -DENABLE_RX_SIGNAL_BAR -DENABLE_TX_TIMEOUT_BAR -DENABLE_COPY_CHAN_TO_VFO_TO_CHAN -DENABLE_SIDE_BUTT_MENU -DENABLE_KEYLOCK  -I c:/1of11/ -I c:/1of11//external/CMSIS_5/CMSIS/Core/Include/ -I c:/1of11//external/CMSIS_5/Device/ARM/ARMCM0/Include -c app/menu.c -o app/menu.o
app/menu.c: In function 'writeXtalFreqCal':
app/menu.c:54:24: error: 'BK4819_REG_3B' undeclared (first use in this function); did you mean 'BK4819_REGS_H'?
   54 |   BK4819_WriteRegister(BK4819_REG_3B, 22656 + value);
      |                        ^~~~~~~~~~~~~
      |                        BK4819_REGS_H
app/menu.c:54:24: note: each undeclared identifier is reported only once for each function it appears in
make: *** [app/menu.o] Błąd 1
Press any key to continue . . .

pawol, 2023-10-25:
doubled

closed 2023-10-25 by pawol

OneOfEleven, 2023-10-25:
ah, you found another :)

Been getting rid of those darned enums, you found one that was hiding !