OneOfEleven/uv-k5-firmware-custom #229: Compnd's C flag is offset off the screen

SylarMa, 2023-10-25:
Compnd's C flag is offset off the screen

In ch-dis, the frequency name will overflow off-screen when it is 10 characters

OneOfEleven, 2023-10-25:
need a picture showing that

SylarMa, 2023-10-25:
[image]
[image]

OneOfEleven, 2023-10-25:
actually the 'C' is fine, that is the what the C looks like in the smallest font.

SylarMa, 2023-10-25:
if the hrb-650~sb becomes shorter,then the STE display will be fine.

SylarMa, 2023-10-25:
The display is already normal, but the c display is not good looking

OneOfEleven, 2023-10-25:
no ?

Looks like a little 'C' to me, they are tiny characters, 3 pixels wide by 5 pixels in height..

SylarMa, 2023-10-25:
ok That looks very cool

closed 2023-10-25 by SylarMa