OneOfEleven/uv-k5-firmware-custom #230: Trim Trailing Zeros

wrcrooks, 2023-10-25:
I believe that the Makefile logic has the inverse effect on the behavior of the radio. When `ENABLE_TRIM_TRAILING_ZEROS := 0` the trailing zeros are trimmed and when `ENABLE_TRIM_TRAILING_ZEROS := 1` the trailing zeros are visible.

OneOfEleven, 2023-10-25:
That can't be ??

OneOfEleven, 2023-10-25:
It's set to '1' for me, the trailing zeros are gone.

OneOfEleven, 2023-10-25:
Post an image if you can.

wrcrooks, 2023-10-25:
Interestingly, I've just compiled it with the option enabled and disabled and the VFO shows the trailing zeros either way.

With it set to '1'
[image]
[image]

With it set to '0'
[image]
[image]

OneOfEleven, 2023-10-25:
ah, i see !

I haven't yet got to removing the trailing zeros from the bug frequency read outs, done the smaller mode ones though.

OneOfEleven, 2023-10-25:
You use big font mode, I should really do that myself on and off to test things out, but never do.

wrcrooks, 2023-10-25:
And now that you say that, I see too :smiley: I noticed the trailing zeros were gone when I merged your changes (including having the BIG_FREQ disabled).

As for testing with big font mode, it's your firmware for your end user experience, you shouldn't have to! Haha

wrcrooks, 2023-10-25:
Just a question based on your experience with the code, does this feature operate such that it doesn't paint the trailing zeros on the UI (thus saving code space) or that it has additional logic to remove the trailing zeros from the UI (thus taking up more FLASH storage)?

OneOfEleven, 2023-10-25:
Try it now with big font

OneOfEleven, 2023-10-25:
It does take a bit more to remove trailing zeros, I have to check each trailing char until I reach a non-zero (going backwards).

wrcrooks, 2023-10-25:
Tested the changes and it works with big font, up to one digit after the decimal (440.000 changes to 440.0)

Thanks for the fix!

closed 2023-10-25 by wrcrooks

OneOfEleven, 2023-10-25:
I could remove the last zero and decimal point but I prefer to keep them

wrcrooks, 2023-10-25:
No objections here, I like it too. 

In my testing this option takes up 44 bytes, in case you want to put that in the README