OneOfEleven/uv-k5-firmware-custom #234: Scan frequency and sub tone errors

SylarMa, 2023-10-26:
When you press and hold 4 or 3 to enter scanning mode, after scanning for frequency and sub tone, all buttons are not working properly and can only be turned off and restarted.

zhaoparrot, 2023-10-26:
do u mean exit key is not working? after u get a result or before u get a scan result? seems working fine on my device.

SylarMa, 2023-10-26:
> do u mean exit key is not working? after u get a result or before u get a scan result? seems working fine on my device.

After ，If there is no result, everything is normal

zhaoparrot, 2023-10-26:
I did a test on F+*  
after i got a result 
exit key chould work but the screen is not displaying frequency
[image]
ignore the RX pixel reverse，that's me trying to make it more noticeable
I try enter menu and exit ,  still no display
long press 3 key to VFO mode it can display alright, but when i press back to MR mode it still not displaying frequency, UP/DOWN key can change 
when I try again I got this 
[image]

zhaoparrot, 2023-10-26:
I tried F+* again and check the menu, found that the CH DIS item on the walkie-talkie displays abnormal content (garbled text). After attempting to modify it, the PonMSG setting has unexpectedly changed.

OneOfEleven, 2023-10-26:
My bad, I'll fix that

OneOfEleven, 2023-10-26:
Don't use F+*  until I've fixed it !!

OneOfEleven, 2023-10-26:
Others have been editing the code recently which I'm not used too, I can back track on everything I do and fix the things I break, but I can't do that with other peoples alterations, that's when I loose track of what's changed.

SylarMa, 2023-10-26:
Don't use 4 FC(F+4),too

SylarMa, 2023-10-26:
I am looking forward to decoding RX MDC1200. Can you do it?

zhaoparrot, 2023-10-26:
> I am looking forward to decoding RX MDC1200. Can you do it?

if u have interest in this topic . iradio5118plus have the ability to display callsign and other stuff, even chinese string . the [magic code is here ](https://github.com/DualTachyon/radtel-rt-890-oefw/blob/main/radio/data.c#L93) it may be using MDC1200.

zhaoparrot, 2023-10-26:
> Others have been editing the code recently which I'm not used too, I can back track on everything I do and fix the things I break, but I can't do that with other peoples alterations, that's when I loose track of what's changed.

While you are looking at this code, when using F+4 or F+*, the current Slist (1 or 2 at the top of the screen) should not be displayed.

SylarMa, 2023-10-26:
> While you are looking at this code, when using F+4 or F+*, the current Slist (1 or 2 at the top of the screen) should not be displayed.

5118p可以接收其它机器的mdc吗？例如摩托，海能达等，

如果可以在k5实现rx mdc，那不是更好吗？

SylarMa, 2023-10-26:
> if u have interest in this topic . iradio5118plus have the ability to display callsign and other stuff, even chinese string . the [magic code is here [...]

5118p可以接收其它机器的mdc吗？例如摩托，海能达等，

如果可以在k5实现rx mdc，那不是更好吗？

OneOfEleven, 2023-10-26:
Safe to use again now, fixed it.

It was causing a lot of sram to be wiped :(

closed 2023-10-26 by SylarMa

zhaoparrot, 2023-10-27:
> Safe to use again now, fixed it.
> 
> It was causing a lot of sram to be wiped :(

I wonder what's the reason of this, is it  UI_PrintString() somehow causeing an overflow?

zhaoparrot, 2023-10-27:
F+* is not displaying slist but u forgot F+4