OneOfEleven/uv-k5-firmware-custom #235: Rx problem in both Vhf 162Mhz and Uhf 462Mhz

Zahurzahur, 2023-10-26:
Hi
I m testing and using different Firmwares on my two K5 wakie talkies from many days and comparing Rx with my Baofeng.
 I have fonud Rx problem in both Vhf 162Mhz and Uhf 462Mhz in this FW. Signal reception is weak then Custom firmware. All other functions are working good and unbeatale.
Sorry if you mind my observation.

OneOfEleven, 2023-10-26:
Some are saying it's deaf, some are saying the opposite, can't keep up

I need people to put there radios onto signal generators and report the actual sensitivity reading at various vhf/uhf frequencies (no out of band reports as they don't count) so as we know either way.

Zahurzahur, 2023-10-26:
Sir
general isue is being faced. I am checking on two k5 radios keeping in same invoirment ans receiving the signals of a transmeter with is hundred miles for away from me.
Its fieldd test. I think its reliable. Ihave chageed different Fw many yimes again when i confermed this isue then I thought to give in your knowlege.

OneOfEleven, 2023-10-26:
oo I don't want any rewards or hardware or anything, but thank you for the offer.

I'm doing it while I can, I don't know how long I can keep going though due to health reasons, time will tell.

Zahurzahur, 2023-10-26:
> oo I don't want any rewards or hardware or anything, but thank you for the offer.
> 
> I'm doing it while I can, I don't know how long I can keep going though due to health reasons, time will tell.

Sir
May God give you health and happiness. Your acheavments are blessings for us.
Sir I have Checked reception on MCFW v29.1C, its Rx on  Band and 70cm  is best ,as equal to iCom V85 walkie Talkie.
Sir I have not much knowledge about softwares , you can find whats happening that is decreasing signals.

bricky149, 2023-10-26:
> I need people to put there radios onto signal generators and report the actual sensitivity reading at various vhf/uhf frequencies (no out of band reports as they don't count) so as we know either way.

By that I'm assuming hardware similar to a NanoVNA. I may get one now that the AGC fiasco has me stumped on this sensitivity issue. I have four K5s, three with their own quirks. Gotta be something between them.

OneOfEleven, 2023-10-26:
Well, if you can figure out the BK4819 chip, well done !!  don't think anyone else has (yet), though some are using registered not listed anywhere, how they do that I do not know.

bricky149, 2023-10-26:
> Well, if you can figure out the BK4819 chip, well done !! don't think anyone else has (yet), though some are using registers not listed anywhere that I'm aware of, how they do that I do not know.

Thought I'd dig deeper and it turns out fagci has found some, including interesting entries like REG_62 (not set).
https://github.com/fagci/uv-k5-firmware-fagci-mod/blob/f71cee19d0f6cfb581f9286a034821fa868066d9/app/spectrum.h#L174

I'm going to play with these and see what happens.

Zahurzahur, 2023-10-28:
You are working great, we are addicted of this.

closed 2023-10-28 by Zahurzahur