OneOfEleven/uv-k5-firmware-custom #239: Inconsistent Firmware Compilation Size

wrcrooks, 2023-10-26:
I was trying to see how much space enabling 'ENABLE_SQUELCH_MORE_SENSITIVE' took up and I noticed that compile size is fluctuating every time I compile (between 62,592 and 62,604 bytes). Sometimes 62,594, sometimes 62,600, sometimes 62,602. Any idea why the project could be outputting different size bins when it stays identical? Is something in the makefile dynamic?

Makefile parameters:
```
ENABLE_CLANG                     := 0
ENABLE_SWD                       := 1
ENABLE_OVERLAY                   := 0
ENABLE_LTO                       := 1
# UART Programming 2.9 kB
ENABLE_UART                      := 1
ENABLE_UART_DEBUG                := 0
# AirCopy 2.5 kB
ENABLE_AIRCOPY                   := 1
ENABLE_AIRCOPY_REMEMBER_FREQ     := 1
ENABLE_AIRCOPY_RX_REBOOT         := 0
# FM Radio 4.2 kB
ENABLE_FMRADIO_76_90             := 0
ENABLE_FMRADIO_68_108            := 1
ENABLE_FMRADIO_76_108            := 0
ENABLE_FMRADIO_875_108           := 0
# NOAA 1.2 kB
ENABLE_NOAA                      := 0
# Voice 1.7 kB
ENABLE_VOICE                     := 0
ENABLE_MUTE_RADIO_FOR_VOICE      := 0
# Tx on Voice 1.0 kB
ENABLE_VOX                       := 0
ENABLE_REDUCE_LOW_MID_TX_POWER   := 1
# Tx Alarm 0.6 kB
ENABLE_ALARM                     := 0
ENABLE_TX1750                    := 0
# MDC1200 0.892 kB
ENABLE_MDC1200                   := 0
ENABLE_PWRON_PASSWORD            := 0
ENABLE_RESET_AES_KEY             := 0
ENABLE_BIG_FREQ                  := 1
ENABLE_SMALL_BOLD                := 0
# trim trailing 0.044 kB
ENABLE_TRIM_TRAILING_ZEROS       := 0
ENABLE_KEEP_MEM_NAME             := 1
ENABLE_WIDE_RX                   := 1
ENABLE_TX_WHEN_AM                := 1
# Freq calibration 0.188 kB
ENABLE_F_CAL_MENU                := 0
ENABLE_TX_UNLOCK                 := 1
ENABLE_CTCSS_TAIL_PHASE_SHIFT    := 1
ENABLE_CONTRAST                  := 0
ENABLE_BOOT_BEEPS                := 0
ENABLE_DTMF_CALL_FLASH_LIGHT     := 1
ENABLE_SHOW_CHARGE_LEVEL         := 1
ENABLE_REVERSE_BAT_SYMBOL        := 1
ENABLE_FREQ_SEARCH_TIMEOUT       := 0
ENABLE_CODE_SEARCH_TIMEOUT       := 0
# Kill and Revive 0.4 kB
ENABLE_KILL_REVIVE               := 0
# AM Fix 0.8 kB
ENABLE_AM_FIX                    := 1
ENABLE_AM_FIX_SHOW_DATA          := 0
# Squelch 0.012 kB .. can't be right ?
ENABLE_SQUELCH_MORE_SENSITIVE    := 0
ENABLE_SQ_OPEN_WITH_UP_DN_BUTTS  := 1
ENABLE_FASTER_CHANNEL_SCAN       := 1
ENABLE_COPY_CHAN_TO_VFO_TO_CHAN  := 1
# Rx Signal Bar 0.4 kB
ENABLE_RX_SIGNAL_BAR             := 0
# Tx Timeout Bar 0.2 kB
ENABLE_TX_TIMEOUT_BAR            := 0
# Tx Audio Bar 0.3 kB
ENABLE_TX_AUDIO_BAR              := 0
# Side Button Menu 0.3 kB
ENABLE_SIDE_BUTT_MENU            := 1
# Key Lock 0.4 kB
ENABLE_KEYLOCK                   := 0
#ENABLE_PANADAPTER               := 0
#ENABLE_SINGLE_VFO_CHAN          := 0
```

OneOfEleven, 2023-10-26:
A mystery indeed !

bricky149, 2023-10-26:
Remove -fmoduo-sched from CFLAGS. I've noticed that can make the bin fluctuate by ~16 bytes and it should've stayed local to my machine as I didn't know how GCC 10.x handled it.

OneOfEleven, 2023-10-26:
Okey doke, I've removed that from the Makefile, I'll upload shortly.

wrcrooks, 2023-10-27:
Tested the fix and the firmware size still fluctuates.
I noticed `-fmoduo-sched` only gets applied when `ENABLE_CLANG := 1`, my Makefile has it set to 0

wrcrooks, 2023-10-27:
gcc version 10.2.1 20210110 (Debian 10.2.1-6)

^ Is that GCC version a problem?

bricky149, 2023-10-27:
> Tested the fix and the firmware size still fluctuates. I noticed `-fmoduo-sched` only gets applied when `ENABLE_CLANG := 1`, my Makefile has it set to 0

So GCC 10.x ignores it. I use GCC 3.x so I can't comment further.

SylarMa, 2023-10-27:
I compiled using Win and did not encounter this issue

wrcrooks, 2023-10-27:
> It's the other option, -freorder-blocks-algorithm=stc. -O2 uses stc, -Os uses simple. The former saves us 100 bytes despite what GCC docs say

Can you elaborate on this?

bricky149, 2023-10-27:
According to the [documentation](https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html#index-O2), -O2 uses `-freorder-blocks-algorithm=stc` when compiling. [Os](https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html#index-Os) disables this, opting for `-freorder-blocks-algorithm=simple` to keep code size down. In our case, stc ([software trace cache](https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html#index-freorder-blocks-algorithm)) reduces the compiled binary by ~100 bytes. I didn't foresee it making the bin size vary between builds.

wrcrooks, 2023-10-27:
So it sounds like we want to keep STC because it reduces the compiled binary by ~100 bytes and close this issue because the fluctuation just seems to be a side-effect of a good compile option?

bricky149, 2023-10-27:
I would agree with that, or at least remove it while counting the size of each feature and putting it back afterwards. I would like to eventually refactor some parts now that I think the AGC stuff is done.

wrcrooks, 2023-10-28:
Roger that. Thanks for the explaination @bricky149 !

closed 2023-10-28 by wrcrooks