OneOfEleven/uv-k5-firmware-custom #240: Can the Reverse frequency function be added to MAKEFILE

SylarMa, 2023-10-26:
Reverse frequency (F+8) is very practical, switching between direct frequencies

I have raised a lot of questions recently, and there are many things I don't understand. I'm very sorry.

SylarMa, 2023-10-26:
I have discovered another issue. When you press the menu button, up and down buttons, etc., the nearby radio will receive a brief signal prompt.

OneOfEleven, 2023-10-26:
Issues get totally lost if you mix them in other threads, you'll have to make a new issue for each issue.

F+8 should work ?

ea7hjb, 2023-10-27:
F+8 in 78da1d5 seems to work well. However (let's see if I can explain this), if you set a frequency in VFO with a TX Offset and move the frequency up or down, the offset symbol will disappear from the screen, and PTTing will not occur with displacement. That is, effectively the offset is disabled. BUT if you go to menu, you will see how the menu retains the offset (showing +/- instead of OFF), and if you press F+8 it will work like if the displacement is enable (and will reverse the frequency). Hope I explained well. I could record a video if I didn't.

closed 2023-10-27 by SylarMa