OneOfEleven/uv-k5-firmware-custom #243: Discussion: Pin Compatible MCU Replacement

wrcrooks, 2023-10-26:
I'm posting this here because this is my favorite modding community for this radio :smile: 

Has anybody tried to look for a pin-compatible replacement for the DP32G030? I was eyeing the STM32G431KBT6 but a few of the pins don't match up. Namely, I think we'd lose PA13 and PA14, but we'd gain PA0 on the STM32 (it wouldn't be wired up to anything on the board). Someone with a steady hand could lift the PA0 pin and restore the functionality of PA13 or PA14 with a small wire across the MCU.

MCU Swap Gains:
- 170 MHz clock over stock 72 MHz
- 128 Kb of flash over stock 64 Kb (More code!!)
- Mathematical hardware accelerators

MCU Swap Losses:
- CRC in Hardware

My main question to @OneOfEleven is: do you know what pins PA13 and PA14 do?

DP32G030:
[image]

STM32G431KBT6:
[image]

wrcrooks, 2023-10-26:
FYI, I know that such a swap to any other MCU would require a major overhaul of this code base, in addition to updates of the HDLs. This is all strictly hypothetical right now, nobody should commit to anything, seriously.

DualTachyon, 2023-10-26:
PA13 at least is used by the voice chip:

https://github.com/DualTachyon/uv-k5-firmware/blob/main/driver/gpio.h

wrcrooks, 2023-10-26:
Sounds like PA13 would have to be migrated to PA0 on the STM32 and a wire taken across to make it work. Again, still hypothetical. What about PA14? If it was needed, that'd make this chip out of contention.

DualTachyon, 2023-10-26:
There were various discussions about this on Telegram, and while there are many MCUs that could fit the bill, it is a fairly big undertaking to bring up the software. It is not hard per se but no one wants to do it.

wrcrooks, 2023-10-26:
> There were various discussions about this on Telegram, and while there are many MCUs that could fit the bill, it is a fairly big undertaking to bring up the software. It is not hard per se but no one wants to do it.

Did they settle on a chip that would work?

DualTachyon, 2023-10-26:
No because without software, it's dead on arrival.

In addition to the voice chip, PA13 is shared with the key pad matrix.

wrcrooks, 2023-10-26:
I'm not volunteering to be the one to port it over to a different MCU **yet** (I've seen how much you and OneOfEleven have been through haha) but I wouldn't mind doing some preliminary testing (chip swap, getting buttons and screen working) if we found a chip.

DualTachyon, 2023-10-26:
Chips from Arterytek might be a better choice because they are easy to buy and cheap, have a good SDK (which I use in the RT-890 OEFW).
 Most of them are STM32 pin compatible (like the dp32).

wrcrooks, 2023-10-26:
> Chips from Arterytek might be a better choice

The Cortex M4 series?
https://www.arterytek.com/en/product/AT32F425.jsp

DualTachyon, 2023-10-26:
A better alternative that was discussed was making our own K5 clone dev kit. A PCB with an AT32 MCU, the bk4819 and better designed RF frontends, etc.

DualTachyon, 2023-10-26:
I don't know the exact model, but yes that series of MCUs.

wrcrooks, 2023-10-26:
> A better alternative that was discussed was making our own K5 clone dev kit. A PCB with an AT32 MCU, the bk4819 and better designed RF frontends, etc.

That would be a massive undertaking but if done, the outcome would be amazing. Have you been able to source the BK4819 chip in the UK?

DualTachyon, 2023-10-26:
In the prototype schematic of the clone, they took an AT32F421 and pin matched it to the K5 connections for LCD, key matrix and BK4819

DualTachyon, 2023-10-26:
No but my wife can buy them in China. They're available on Taobao.

wrcrooks, 2023-10-26:
I have so many questions, sorry haha. This work on a prototype clone, is this all being done in Telegram?

DualTachyon, 2023-10-26:
The clone would be easier to do the SW coz I can just take the RT-890 firmware and strip it down. I don't have to figure out how to make the base BSP work.

Someone already replaced the MCU in the 890 with the "same" MCU bought from Ali express, and the 890 works the same.

DualTachyon, 2023-10-26:
There is no prototype yet, just some PCB design (single person on part time).

He plans to make a zip file that you can upload to JLCpcb, and they will make the PCB and solder all the components for you for cheap. I don't know the price exactly, but like 10-30$us max?

wrcrooks, 2023-10-26:
I mean, if you were going to design a complete replacement board, you could just go for broke and put in USB-C PD, a few megabytes of FLASH, a much larger EEPROM, a dual core MCU, more hardware offloading. The whole thing.

DualTachyon, 2023-10-26:
That's the plan, kind off, but not replacing it on the K5. Just have a PC tethered radio board that we can experiment with RF tx/rx

DualTachyon, 2023-10-26:
Making a design that mimics the K5 layout makes it easier to debug and bring up.

DualTachyon, 2023-10-26:
This was the last I saw of it (I'm not the one doing it).

[image]

wrcrooks, 2023-10-26:
This is amazing stuff... How feasible do you think it would be to put in enough RF filtering to allow for TX'ing on all amateur radio bands inside of 18-1200MHz? I believe it'd have to have one filtered path for each band, right? That's a lot of PCB space.

DualTachyon, 2023-10-26:
I don't know. I think quality RF means both expensive and bigger PCB area, based on discussions I had before.

DualTachyon, 2023-10-26:
The designer of that layout knows RF pretty well too.

wrcrooks, 2023-10-26:
> I can just take the RT-890 firmware and strip it down

If you don't mind my asking, which parts of your RT-890 firmware would have to be stripped out? I was unfamiliar with that radio until you mentioned it so I'm not sure where their specs differ.

DualTachyon, 2023-10-26:
Mostly components not attached, like the BK1080 FM receiver.

DualTachyon, 2023-10-26:
The 890 is what I call a "color K5". It has the AT32 I mentioned,a color screen, 4MB of SPI flash and the rsame BK radio chips. Check my profile for the OEFW for it. I have the K5 FW running on the 890. (Took 3 hours of porting).

wrcrooks, 2023-10-26:
You are a legend! 

> 4MB of SPI flash

This alone would make the radio's extra $20 price tag worth it. Does the color screen add to the size of the code, given the extra color attributes for all UI elements?

wrcrooks, 2023-10-26:
Sounds like I need to get an RT-890 and get on Telegram to keep up with all the progress being made in this community.

DualTachyon, 2023-10-26:
Progress on the 890 is not as good because it's 99.99% people waiting for ultraHD-1of11-fagci-merge binaries without even helping test base OEFW (I might still have a few bug). So don't count on community stuff to happen for now. There are only a few people doing stuff and only part time.

The color screen is more of an issue for performance, because 1 pixel is 2 bytes while K5 1 byte is 8 pixels. This goes over SPI, which is bit banged instead of native SPI hardware. However, I ran fagci's spectrum on 890 (not released) and I don't see any performance issues.

890 runs at 72mhz, while K5 runs 48mhz. I also added a comment to enable 120mhz and I'm told the radio works fine.

DualTachyon, 2023-10-26:
...

https://filetransfer.io/data-package/3yPhwwLq#link

wrcrooks, 2023-10-26:
Ooo, overclocking! Yes sir, it is settled. RT-890 has been purchased. Once I receive, I'll do any testing I can to help. I appreciate you staying up to answer all these random questions.

wrcrooks, 2023-10-26:
> https://filetransfer.io/data-package/3yPhwwLq#link

Interesting, you're painting the UV-K5 UI in black and white in the center of the colored screen because there's a pixel density difference and you haven't gotten around to adding color to the UI painting commands yet? That's a very impressive feat for 3 hours.

DualTachyon, 2023-10-26:
Colorising the K5 is not easy because of the way it does rendering. Everything is placed on a 128x64 in-memory frame buffer and then all sent in one go. Color information is lost. I could add a color buffer for each pixel and then change colours around draw calls, but not that interesting to me.

On the 890, pixels are sent immediately to the LCD as things are drawn. Its screen is also bigger. 160x120 or so. Physically it is even bigger, but almost 1/3 of it is under the case.

wrcrooks, 2023-10-26:
Very cool stuff, I can't wait to dig into it when my 890 arrives. I'd argue color is nice to have but I'm not sure it's really necessary for a simple handheld radio, especially if cosmetic code takes up space that could go towards functionality code.

zhaoparrot, 2023-10-26:
the iradio 5118 plus(RT-890）also support full chinese character display ,which requires a big chunk of flash

DualTachyon, 2023-10-26:
I saw some Chinese font in the English 890 already but I don't know if it's full character set.

There's around 2 Mbytes of voice samples that could be repurposed since they slow down everything, like on the K5.

wrcrooks, 2023-10-26:
And with open source firmware on it, I'm sure you could choose one or the other to clear up some space. I was a big fan of being able to compile without voice.

zhaoparrot, 2023-10-26:
some local ham on repeater said that 5118plus can transfer callsign or even a short chinese string at bot or eot, I suppose 890 can also do that, right?

DualTachyon, 2023-10-26:
Yes. It's roger beep mode 3. It sends an ID and you can edit it via the menu.

DualTachyon, 2023-10-26:
https://github.com/DualTachyon/radtel-rt-890-oefw/blob/main/app/radio.c#L497

DualTachyon, 2023-10-26:
In 890 case it seems to be EOT only.

wrcrooks, 2023-10-26:
Is that along the lines of the MDC1200 implementation that OneOfEleven implemented?

DualTachyon, 2023-10-26:
11 thought that it was earlier, but it's lacking the CRC stuf and other protocol bits. But the base code might work for MDC1200 if you port over the protocol 11 wrote.

zhaoparrot, 2023-10-27:
> In 890 case it seems to be EOT only.

thanks for the link ,it chould also be EOT only in 5118plus too. I'm not sure.

wrcrooks, 2023-10-28:
This was a good conversation. Thanks to everyone!

closed 2023-10-28 by wrcrooks