OneOfEleven/uv-k5-firmware-custom #245: Weird noises on receive related to BK4819 register 7E

egzumer, 2023-10-26:
You set AGC fix mode to 1 in BK4819_DisableAGC, it is called only once on startup. I see 2 bugs here. 
One, I get buzzing noises on receive until I do some transmission, and here comes number two. The first write to reg 7E is overwritten in BK4819_ExitBypass which permanently negates the first call and sets AGC fix mode to 0, which in turn fixes the buzzing.

bricky149, 2023-10-27:
> You set AGC fix mode to 1 in BK4819_DisableAGC, it is called only once on startup. I see 2 bugs here. One, I get buzzing noises on receive until I do some transmission, and here comes number two. [...]

I'll respond to this as the changes relate to me.

When I first proposed the addition of writing to REG_7E, I saw no other reference to it at the time. I thought 1 enabled and 0 disabled AGC but it's the opposite; 0 sets to auto control, 1 sets to fixed control. The decision to use fixed control was for airband although I'm aware AGC affects Tx as well.

This issue confirms AGC does something after all. My discovery of an undocumented register may yield results. For now, I ask you and 1o11 to leave AGC stuff with me since I'm obsessing over it in the hopes it can actually help us.

egzumer, 2023-10-27:
Yeah, but it was ALWAYS enabled then. It is set to 0 in BK4819_ExitBypass.

bricky149, 2023-10-27:
I didn't notice BK4819_ExitBypass until you brought it up. AFAIK that was never touched. I'm acutely aware of the buzzing you mean, please give me time to understand why this is. All I can say is there must be a reason why stock firmware had an if-else branch in SetAGC, we were only using one or the other but never switching between them.

egzumer, 2023-10-27:
But REG_7E wasn't ever written to in the SetAGC, this is something that was only introduced in recent days. The fix by fagci that I believe it all started with was all about REG_06.

bricky149, 2023-10-27:
I put the reg write there because it related to AGC. REG_7E, as previously stated, controls whether gain control is fixed (1) or automatic (0). REG_10, 11, 12, 13 and 14 are the tables used for gain control, each controlling LNAS, LNA, Mixer and PGA.

As I've also stated, I was unaware ExitBypass wrote to REG_7E by toggling AGC on. AGC appears to affect Tx so it makes sense it would be used there. The decision to disable AGC on startup wasn't my own, however I understand why it was done. It is clear it has led to side-effects.

I am interested in AGC because I want to know if it can resolve the AM distortion. I have found out there's an undocumented register that may tell AGC what we want to limit LNA to, in this case I'm using 143 (default is 255). The hope is I can address this while also addressing the sensitivity issues others are mentioning.

I appreciate your patience on this.

bricky149, 2023-10-27:
I think I've worked it out. Remarks about what I was doing may be found in #244.

1o11 has disabled the ExitBypass function so AGC is perma-off as intended. If my changes are picked up, and AGC is turned back on, it may be worth looking at setting different thresholds between AM and FM bands since REG_62 impacts AGC behaviour. I was only looking at airband and that's probably why AGC was disabled entirely.

OneOfEleven, 2023-10-28:
Is this noise still happening I wonder ?

egzumer, 2023-10-28:
No more buzzing with last version on my radio

closed 2023-10-28 by OneOfEleven