OneOfEleven/uv-k5-firmware-custom #248: Request:develop RX MDC

SylarMa, 2023-10-27:
The completion of TX MDC function is really great.

TX MDC has been developed. Can you develop RX MDC?

I believe that RX MDC has many people looking forward to it.

wrcrooks, 2023-10-27:
OneOfEleven stated in #173 that he's leaving decoding of MDC to someone else

SylarMa, 2023-10-27:
> OneOfEleven在#173中表示，他将MDC的解码留给其他人

Maybe it was told to me at the time, it might have been a joke.
If it's true, that's too bad!

OneOfEleven, 2023-10-28:
The problem (a big one) with doing MDC receive is the the K5 radio chip requires a short length of bit toggling pre-amble at the start of the packet, the pre-amble allows receivers to sync their bit clock PLL's.

But, motorola made a mistake when they created MDC, the designer never added that much needed bit toggling pre-amble, which means the K5 BK4810 FSK modem part can't sync it's bit clock at the start of each and every packet, which means a lot of missed packets.

Most all other packet transmission standards start their packets with this much needed bit toggling pre-amble sequence, motorola MDC don't. It can be made to work **IF** the FSK modem demodulator is designed to work with MDC, but the K5's BK4819 chip has not been so designed.

The other problem now is CPU flash space.

closed 2023-10-28 by OneOfEleven

SylarMa, 2023-10-28:
> 进行 MDC 接收的问题（一个大问题）是 K5 无线电芯片在数据包开始时需要一小段位切换前导码，前导码允许接收器同步其位时钟 PLL。
> 
> 但是，摩托罗拉在创建 MDC 时犯了一个错误，设计师从未添加急需的位切换前导码，这意味着 K5 BK4810 FSK 调制解调器部分无法在每个数据包的开头同步其位时钟，这意味着很多丢失的数据包。
> 
> 大多数其他数据包传输标准都以这个急需的位切换前导码序列开始其数据包，摩托罗拉 MDC 则没有。**如果**FSK调制解调器设计为与MDC一起使用，则可以使其工作，但K5的BK4819芯片尚未如此设计。
> 
> 现在的另一个问题是CPU闪存空间。

I was very excited to know that k5 can TX MDC.
But now I know that K5 cannot RX MDC, which is very frustrating.
TX MDC has become meaningless.
Bye k5!

OneOfEleven, 2023-10-28:
Gee thanks for all the past weeks work I've done for you,

oh it's no problem at all, your welcome.

Bye you !

493505110, 2023-10-28:
Is it possible to replace BK4819 chip with other chip?

DualTachyon, 2023-10-28:
> Is it possible to replace BK4819 chip with other chip?

Give it a try and let us know the results.