OneOfEleven/uv-k5-firmware-custom #249: Cannot compile the latest commit with uart debug disabled

pawol, 2023-10-27:
ntic -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"ab0c372\" -DENABLE_FMRADIO_875_108 -DENABLE_FMRADIO -DENABLE_UART -DENABLE_SMALL_BOLD -DENABLE_MUTE_RADIO_FOR_VOICE -DENABLE_VOX -DENABLE_REDUCE_LOW_MID_TX_POWER -DENABLE_TX1750 -DENABLE_MDC1200 -DENABLE_RESET_AES_KEY -DENABLE_KEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_F_CAL_MENU -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_AM_FIX -DENABLE_SQUELCH_MORE_SENSITIVE -DENABLE_FASTER_CHANNEL_SCAN -DENABLE_RX_SIGNAL_BAR -DENABLE_TX_AUDIO_BAR -DENABLE_COPY_CHAN_TO_VFO_TO_CHAN -DENABLE_SIDE_BUTT_MENU -DENABLE_KEYLOCK  -I c:/1of11/ -I c:/1of11//external/CMSIS_5/CMSIS/Core/Include/ -I c:/1of11//external/CMSIS_5/Device/ARM/ARMCM0/Include -c main.c -o main.o
main.c: In function '_putchar':
main.c:53:2: error: implicit declaration of function 'UART_Send' [-Werror=implicit-function-declaration]
   53 |  UART_Send((uint8_t *)&c, 1);
      |  ^~~~~~~~~
main.c: In function 'Main':
main.c:74:2: error: implicit declaration of function 'UART_Init'; did you mean 'BOARD_Init'? [-Werror=implicit-function-declaration]
   74 |  UART_Init();
      |  ^~~~~~~~~
      |  BOARD_Init
main.c:79:3: error: implicit declaration of function 'UART_SendText' [-Werror=implicit-function-declaration]
   79 |   UART_SendText(UART_Version_str);
      |   ^~~~~~~~~~~~~
cc1.exe: all warnings being treated as errors

SylarMa, 2023-10-27:
ENABLE_UART                      := 1
ENABLE_UART_DEBUG                := 1

Both must be turned on simultaneously, and I noticed that the firmware will become larger.

pawol, 2023-10-27:
Yes, I noticed that FW is bigger also, Same option as always but the size is oversized 700 bytes then previously. I never used uart debug option - always was 0

SylarMa, 2023-10-27:
Yes, so I have turned off other functions. The latest update requires both options to be turned on simultaneously.

wrcrooks, 2023-10-27:
The latest commit ab0c372 uses
`if defined(ENABLE_UART) && defined(ENABLE_UART_DEBUG)`

I don't want to assume the intention but maybe that was supposed to be or (||) instead of and (&&)?

pawol, 2023-10-27:
If needed then the latest commit can be avoided by git checkout 78da1d5

OneOfEleven, 2023-10-27:
fixed it

closed 2023-10-27 by OneOfEleven

wrcrooks, 2023-10-27:
I was still having issues compiling. My fix was to replace 
```
#if defined(ENABLE_UART) && defined(ENABLE_UART_DEBUG)
	#include "driver/uart.h"
```
with
```
#if defined(ENABLE_UART) || defined(ENABLE_UART_DEBUG)
	#include "driver/uart.h"
```
in the following files:
audio.c
frequencies.c
functions.c
main.c
mdc1200.c
settings.c
app/action.c
app/aircopy.c
app/app.c
app/fm.c
app/generic.c
app/main.c
app/menu.c
app/search.c
ui/menu.c

OneOfEleven, 2023-10-27:
show me the errors

wrcrooks, 2023-10-27:
```
main.c: In function 'Main':
main.c:69:2: error: implicit declaration of function 'UART_Init'; did you mean 'BOARD_Init'? [-Werror=implicit-function-declaration]
  UART_Init();
  ^~~~~~~~~
  BOARD_Init
main.c:74:3: error: implicit declaration of function 'UART_SendText' [-Werror=implicit-function-declaration]
   UART_SendText(UART_Version_str);
   ^~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile:508: main.o] Error 1
```

uart.c isn't being loaded with:
```
# UART Programming 2.9 kB
ENABLE_UART                      := 1
ENABLE_UART_DEBUG                := 0
```

wrcrooks, 2023-10-27:
25fee55 did it, thanks!