OneOfEleven/uv-k5-firmware-custom #253: Changing SQL value varies frequency.

ea7hjb, 2023-10-27:
Here goes one of those weird as f**k bugs.

If you copy to VFO a frecuency from memory holding the M key (I only observed this behaviour if this is done as first step!), and then change SQL value, the VFO frequency changes. This happens, in both VFO.

As always, thanks for your effort in advance. Best regards.

SylarMa, 2023-10-28:
I also encountered this problem
The problem is this
1. VFO mode frequency 438.500
2. MR mode, press M, copy frequency 438.600 to VFO
3. Modify SQL
4. The frequency has changed from 438.600 to 438.500

OneOfEleven, 2023-10-28:
That sequence can't be done, you can't copy the freq into the VFO once you go into channel mode. You can only copy the channel freq into the VFO.

OneOfEleven, 2023-10-28:
I can't reproduce this for some reason, don't suppose you could do a little video of it ?

ea7hjb, 2023-10-28:
> I can't reproduce this for some reason, don't suppose you could do a little video of it ?

Sure thing!

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/17039268/0f6fb96e-1425-4d7b-bf57-d435d0ddbda8

Tbh, I don't know how I manage to find these bugs... lol

OneOfEleven, 2023-10-28:
ok, I think it's putting the original VFO frequency back that was there before you did the channel-vfo copy by the looks of it.

Just need to fix it now.

Videos tell me what I need to know, words usually don't :)

ea7hjb, 2023-10-28:
> ok, I think it's putting the original VFO frequency back that was there before you did the channel-vfo copy by the looks of it.

It's not exactly the previous frequency, but the previous frequency of THAT VFO band. In the case of the video, Band 6.

> Videos tell me what I need to know, words usually don't :)

Specially with me poor english, lol.

OneOfEleven, 2023-10-28:
Try it now, I've probably broke it but hopefully not.

THE most confusing thing I've found with how the QS firmware is written is all the different channel/frequency/vfo variables that have to be changed and match up. It's a complete nightmare and has caused lots of problems for me. If the original author goes away then comes back years later, they too will see just how bad their coding was.

ea7hjb, 2023-10-28:
> Try it now, I've probably broke it but hopefully not.
Yes! It looks like it is working well now! Thank you.
 
> THE most confusing thing I've found with how the QS firmware is written is all the different channel/frequency/vfo variables that have to be changed and match up. It's a complete nightmare and has [...]
Perhaps the code was written by an AI? Think of it, lol

And sorry for the off-topic, let me ask you a question, out of pure curiosity (there is nothing wrong with it): why did you change the ALARM sound? Any technical reason, or simply because you liked the new sound more?

OneOfEleven, 2023-10-28:
I only changed it from saw tooth sweep to triangle sweep to see how it sounded, I can change it back.

OneOfEleven, 2023-10-28:
I can even get the radio to play musical notes, up to 2 tones at once, so a 2-tone cord

ea7hjb, 2023-10-28:
Oh no, don't worry. I don't think many people will be affected by this, so do as you like, of course. I just asked just for pure curiosity, nothing else.

OneOfEleven, 2023-10-28:
I'm thinking the ET tune might be novel ;)

closed 2023-10-28 by ea7hjb