OneOfEleven/uv-k5-firmware-custom #254: bye k5！

SylarMa, 2023-10-28:
I was very excited to know that k5 can TX MDC.
But now I know that K5 cannot RX MDC, which is very frustrating.
TX MDC has become meaningless.
Bye k5!
 #248

DualTachyon, 2023-10-28:
Bye and don't come back.

OneOfEleven, 2023-10-28:
I'm sure I left a reply to it here myself, but it's vanished :(

DualTachyon, 2023-10-28:
He came to my 890 repo asking about MDC and sent him away I'm not doing his job

OneOfEleven, 2023-10-28:
Good for you  :)

Smtoo, 2023-10-28:
I am sylarma, and I think it's necessary for me to explain this issue.
So I registered a new account,
My expression above is very incorrect. There is a problem with my translation software and I mistakenly translated what I wanted to say.
Thank you for your selfless work, which has enabled k5 to have a real MDC1200 and the ability to send MDC is also great.
I am deeply sorry for the misunderstanding above.

Smtoo, 2023-10-28:
When I saw the error, I was no longer able to make edits.

DualTachyon, 2023-10-28:
Cool story, bro.

This isn't an airport, you don't need to advertise you're leaving. Such a message often translates into emotional blackmail to force developers to do things against their wishes.

Smtoo, 2023-10-28:
> 很酷的故事，兄弟。
> 
> 这不是机场，你不需要宣传你要离开。这样的信息通常会转化为情感勒索，迫使开发人员做违背他们意愿的事情。

Yes, I am aware of this issue and hope it will not affect your development enthusiasm. I greatly respect your efforts

Smtoo, 2023-10-28:
I want to express that I need to find another radio that supports MDC

Smtoo, 2023-10-28:
Although I cannot edit the program, I also try my best to make my own efforts, test hard, and submit errors.

Smtoo, 2023-10-28:
If you can forgive me, I hope to restore my account. If you don't forgive me, I also hope you don't give up on K5.
You have given k5 new life, and many people are paying attention to k5.

wrcrooks, 2023-10-28:
Leave it alone. Take a break @Smtoo. **Close your outstanding issues**. You've made your apology. Let the developer relax and you can watch his progress without speaking up. He's done a great job in/for this community and you're being divisive.

closed 2023-10-28 by OneOfEleven