OneOfEleven/uv-k5-firmware-custom #258: STE not working

egzumer, 2023-10-28:
These two commits brake STE. You assume tail tone should be emitted only if TX CTCSS is turned on, which is incorrect.
The QS way is, it emits DCS CTC134 tail tone if TX DCS is turned on and CTCSS 55Hz tone otherwise (even when DCS/CSTCSS/STE is all turned off).

Also the faze shift tail tone didn't work either last time I checked.

BTW do you remember what was the issue with "200ms TX tail bug"?

fde690a74aa6730e85cffe4e30076f0faabbed21: app.c
50a55e34ab8bf18ec29b9ae8674ec68d81107afe: radio.c

OneOfEleven, 2023-10-28:
>The QS way is, it emits DCS CTC134 tail tone if TX DCS is turned on and CTCSS 55Hz tone otherwise (even when DCS/CSTCSS/STE is all turned off).

The radio can't transmit CDCSS AND CTCSS at the same time, only one or the other.

The phase shift works perfectly, see the 180deg shift in the CTCSS tone at the end ?  ..

[image]

egzumer, 2023-10-28:
> The radio can't transmit CDCSS AND CTCSS at the same time, only one or the other.

Probably that is why QS uses DCS tail when TX DCS is on, and only then.

> The phase shift works perfectly, see the 180deg shift in the CTCSS tone at the end ? ..

I may well be transmitting it, but it wasn't recognizing it last time I've checked.

OneOfEleven, 2023-10-28:
The 200ms bug (months ago) was holding the TX on for 200ms after the PTT was released even when both CTCSS and CDCSS was turned off.

OneOfEleven, 2023-10-28:
At the time I did the 180deg phase shift was for the repeaters and all other equipment working with CTCSS, I never bothered with the K5 receive side using it. I didn't have the knowledge I now have about the code and BK chip.

I'll look at it at some point, but right now I'm reducing my input to my firmware, I need to move on soon.

egzumer, 2023-10-28:
That is fine, just reporting what I found. Take a look at that when you have time. I reverted back to the original code already.