OneOfEleven/uv-k5-firmware-custom #259: Incorrect RSSI register read

xawen, 2023-10-28:
We found a potential issue in the RT-890 OEFW code that appears to also affect the K5.

It looks like the BK4819_GetRSSI() function is reading what's supposed to be an 8 bit register as 9 bits.  The current read is done as: 
BK4819_ReadRegister(0x67) & 0x01FF;

This is causing the returned RSSI value to be high.  Which explains why most traffic reads as S9+ on the signal bar.

After playing around on the 890, the correct way to read the register seems to be 8 bits, excluding the least significant (code example below).  On the 890, this has produced more sensible dBm values:
- -117 instead of -74 for a repeater several miles away  
- -81 instead of -47 for a 5w transmission with radios side-by-side 

This is the modified read function that I tested on the 890 that seems to return sane values:

uint16_t BK4819_GetRSSI(void)
{
	uint16_t RawRSSI;

	RawRSSI = BK4819_ReadRegister(0x67) & 0x01FE;
	RawRSSI = RawRSSI >> 1;
	return RawRSSI;
}

DualTachyon, 2023-10-28:
Let me check the code in Ghidra for both radios to ensure I didn't write it wrong.

egzumer, 2023-10-28:
Does it really matter how many bits we read? The new code basically returns the same value but divided by 2. From what I know RSSI is just a number that is related to a signal strength. There is no one RSSI to dBm equation if I'm not mistaken.

OneOfEleven, 2023-10-28:
dB = (RSSI / 2) - 160

DualTachyon, 2023-10-28:
I confirmed that 0x1FF is the correct mask used in both radios. On the K5 at least, the value is used against a calibration table to determine the level. They don't use dBm units or anything like that for display.

egzumer, 2023-10-28:
> dBm = (RSSI / 2) - 160

Where does it come from?

xawen, 2023-10-28:
> Does it really matter how many bits we read? The new code basically returns the same value but divided by 2. From what I know RSSI is just a number that is related to a signal strength. There is no one RSSI to dBm equation if I'm not mistaken. We need someone to do the measurements to get RSSI to dBm relation across spectrum.

There's two impacts that I see:
1) The K5 (and 890) always came back with much higher dBm values than my other radios
2) The values get weird when the raw number gets above 256 (8 bits)

On the 890, the high values were also messing up the signal meter, but that's not a K5 problem.

OneOfEleven, 2023-10-28:
It is a 9 bit value, it goes in 0.5dB steps

DualTachyon, 2023-10-28:
The register datasheet shows the formula to be what 11 wrote above. It's nothing new/hidden.

xawen, 2023-10-28:
> The register datasheet shows the formula to be what 11 wrote above. It's nothing new/hidden.

I don't have the sheet, but doesn't it list that as an 8-bit register?

OneOfEleven, 2023-10-28:
no, 9-bit

OneOfEleven, 2023-10-28:
even in your code you read 9 bits but discard the LSB (the 0.5dB bit)

egzumer, 2023-10-28:
[image]

You're right, I didn't check it before, but the original function follows the docs.

xawen, 2023-10-28:
> no, 9-bit

Well...hmm.  Ok, back to the 890 to figure out why the returned values are off the charts for stronger signals when using the 9-bit read.

egzumer, 2023-10-28:
I had someone complain, and I think it is valid, you get S7-S9 in monitoring mode where there is no signal. That doesn't seem right. Also QS for its small s-meter use different calibration for bands 1-2 and 3-7

bands 3,4,5,6,7
lvl 1: 110
lvl 2: 120
lvl 3: 130
lvl 4: 140
bands 1,2
lvl 1: 180
lvl 2: 190
lvl 3: 200
lvl 4: 210

I think our own calibration based on real measurements would be the best.

OneOfEleven, 2023-10-28:
The RSSI reading will be effected by every gain setting and any other gain/atten in the chain from antenna to RSSI reading. It all has to accounted for if you want a final real dBm value.

OneOfEleven, 2023-10-28:
So the **- 160** value needs to be calibrated/varied across the two RF bands taking into account any RF front end filtering etc.

xawen, 2023-10-28:
Closing as a non-issue.

closed 2023-10-28 by xawen