OneOfEleven/uv-k5-firmware-custom #260: High Pitched Noise

mjwgeek, 2023-10-28:
I've noticed that with this firmware, the longer the transmission, the more noticeable a high pitched squealing noise becomes.  I have two of these radios, and I've tried the UVMOD version, stock, and this one, and this is the only one with this issue.  I'm not sure what would cause this issue.  Has anyone else noticed this?

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/83971391/fed16d8c-f380-4537-9dc9-ce7d8fdac84c

OneOfEleven, 2023-10-28:
Swap the position of the radios over, see if the noise stays with one radio, if it does then I guess you have a poorly radio :(

mjwgeek, 2023-10-28:
I've loaded different firmware on each, and it happens on both with this firmware, but it doesn't happen on either with stock or any of the firmware's I mentioned.  It's not the radio.  I've tried different settings in the makefile to see if I could get it to go away and it doesn't.

OneOfEleven, 2023-10-28:
Really have no idea sorry.

bricky149, 2023-10-28:
> I've noticed that with this firmware, the longer the transmission, the more noticeable a high pitched squealing noise becomes. I have two of these radios, and I've tried the UVMOD version, stock, and this one, and this is the only one with this issue. I'm not sure what would cause this issue. Has anyone else noticed this?
> K5.Noise.1.mp4

May I ask what versions these radios are running, and what frequency? I cannot the read the screens.

mjwgeek, 2023-10-28:
The latest version downloaded from git clone here makes the noise on rx on either radio.  Uploaded and added .zip to end, delete to flash). 
[firmware.packed.bin.zip](https://github.com/OneOfEleven/uv-k5-firmware-custom/files/13196539/firmware.packed.bin.zip)

Here is a copy of a firmware that I can flash to either radio that doesn't have the issue.  Again, just remove the .zip at the end.
[fw_modded.bin.zip](https://github.com/OneOfEleven/uv-k5-firmware-custom/files/13196542/fw_modded.bin.zip)

bricky149, 2023-10-28:
> The latest version downloaded from git clone here makes the noise on rx on either radio. Uploaded and added .zip to end, delete to flash). [firmware.packed.bin.zip](https://github.com/OneOfEleven/uv-k5-firmware-custom/files/13196539/firmware.packed.bin.zip)
> 
> Here is a copy of a firmware that I can flash to either radio that doesn't have the issue. Again, just remove the .zip at the end. [fw_modded.bin.zip](https://github.com/OneOfEleven/uv-k5-firmware-custom/files/13196542/fw_modded.bin.zip)
> 
> It happens on any frequency I've tried which I've tried the NOAA frequencies, and both 2Meter and 70cm ham frequencies since I have active repeaters in those ranges.

Could you upload the firmware bin that isn't packed? The packed file is obfuscated.

mjwgeek, 2023-10-28:
Here you go!
[firmware.bin.zip](https://github.com/OneOfEleven/uv-k5-firmware-custom/files/13196599/firmware.bin.zip)

bricky149, 2023-10-28:
Thanks. Your version is 9eb3f765e060707fd5fae1b77c49fc1125d0dfb6.

There are a lot of changes between stock (or OEFW) and this version. Are you able to try a different custom firmware so we know it's this particular version? I can't reproduce your issue on my radios.

bricky149, 2023-10-28:
I think I found your issue.

I have two radios, one with the AM fix and one without. The noise you're hearing is the one I'm hearing from the 'AM fix' radio. This is significant because AGC is disabled when compiled with AM_FIX := 1.

I'll look at it and see what I can do.

mjwgeek, 2023-10-28:
I'll try to go back a few versions and report back.

Ok so, the problem isn't present in version 78bc93, but is in the latest daa3d29.  I can try more versions if you want.

mjwgeek, 2023-10-28:
I compiled the latest with am fix = 0 and it didn't make any difference for me, as in, the noise is still there. 

ENABLE_CLANG                     := 0
ENABLE_SWD                       := 0
ENABLE_OVERLAY                   := 0
ENABLE_LTO                       := 1
# UART Programming 2.9 kB
ENABLE_UART                      := 1
ENABLE_UART_DEBUG                := 0
# AirCopy 2.5 kB
ENABLE_AIRCOPY                   := 0
ENABLE_AIRCOPY_REMEMBER_FREQ     := 0
ENABLE_AIRCOPY_RX_REBOOT         := 0
# FM Radio 4.2 kB
ENABLE_FMRADIO_64_76             := 0
ENABLE_FMRADIO_76_90             := 0
ENABLE_FMRADIO_76_108            := 1
ENABLE_FMRADIO_875_108           := 0
# NOAA 1.2 kB
ENABLE_NOAA                      := 0
# Voice 1.7 kB
ENABLE_VOICE                     := 0
ENABLE_MUTE_RADIO_FOR_VOICE      := 0
# Tx on Voice 1.0 kB
ENABLE_VOX                       := 0
ENABLE_REDUCE_LOW_MID_TX_POWER   := 0
# Tx Alarm 0.6 kB
ENABLE_ALARM                     := 0
ENABLE_TX1750                    := 0
# MDC1200 2.8 kB
ENABLE_MDC1200                   := 1
ENABLE_PWRON_PASSWORD            := 0
ENABLE_RESET_AES_KEY             := 0
ENABLE_BIG_FREQ                  := 0
ENABLE_SMALL_BOLD                := 0
# trim trailing 0.044 kB
ENABLE_TRIM_TRAILING_ZEROS       := 1
ENABLE_KEEP_MEM_NAME             := 1
ENABLE_WIDE_RX                   := 1
ENABLE_TX_WHEN_AM                := 1
# Freq calibration 0.188 kB
ENABLE_F_CAL_MENU                := 0
ENABLE_TX_UNLOCK                 := 0
ENABLE_CTCSS_TAIL_PHASE_SHIFT    := 1
ENABLE_CONTRAST                  := 1
ENABLE_BOOT_BEEPS                := 0
ENABLE_DTMF_CALL_FLASH_LIGHT     := 0
ENABLE_SHOW_CHARGE_LEVEL         := 0
ENABLE_REVERSE_BAT_SYMBOL        := 0
ENABLE_FREQ_SEARCH_TIMEOUT       := 0
ENABLE_CODE_SEARCH_TIMEOUT       := 0
# Kill and Revive 0.4 kB
ENABLE_KILL_REVIVE               := 0
# AM Fix 0.8 kB
ENABLE_AM_FIX                    := 0
ENABLE_AM_FIX_SHOW_DATA          := 0
# Squelch 0.012 kB .. can't be right ?
ENABLE_SQUELCH_MORE_SENSITIVE    := 1
ENABLE_SQ_OPEN_WITH_UP_DN_BUTTS  := 0
ENABLE_FASTER_CHANNEL_SCAN       := 1
ENABLE_COPY_CHAN_TO_VFO_TO_CHAN  := 0
# Rx Signal Bar 0.4 kB
ENABLE_RX_SIGNAL_BAR             := 1
# Tx Timeout Bar 0.2 kB
ENABLE_TX_TIMEOUT_BAR            := 0
# Tx Audio Bar 0.3 kB
ENABLE_TX_AUDIO_BAR              := 1
# Side Button Menu 0.3 kB
ENABLE_SIDE_BUTT_MENU            := 1
# Key Lock 0.4 kB
ENABLE_KEYLOCK                   := 0
#ENABLE_PANADAPTER               := 0
#ENABLE_SINGLE_VFO_CHAN          := 0

bricky149, 2023-10-28:
I've pushed a fix on my issue-260 branch on my repo. Could you compile and flash that to make sure I've found the issue?

It's related to AGC (that old chestnut). I'm not happy that I had to enable it in a func meant to disable it, all I could do was make it as weak as possible.

mjwgeek, 2023-10-28:
Sure thing!

mjwgeek, 2023-10-28:
I complied and flashed.  I'm still experiencing the same issue.  I've noticed that if I unscrew the antenna, in effect weaking the signal, it goes away.  The RSSI and S Meter goes from an S9+46 with the antenna and drops to an s9+24 without, maybe that gives some hint as to what's going on.  On a really strong signal, it occurs, but I don't know what the exact threshold is, but it's not present below S9+24.  And yes I've tried other antennas just to rule out a faulty antenna.

OneOfEleven, 2023-10-30:
Have you got the compander on by any chance ?

mjwgeek, 2023-10-30:
I do not.  I checked that to be sure.  It's just really strong signals so I think it's related to the AGC somehow.

OneOfEleven, 2023-10-30:
Can you do another video like the one you did, but one that shows the screens in detail (can't see anything in your first video) at all ? .. can maybe spot something from

bricky149, 2023-10-30:
> I do not. I checked that to be sure. It's just really strong signals so I think it's related to the AGC somehow.

I was able to isolate the problem in my OEFW fork, it is related to AGC. I would propose exposing its 'fix index' as a setting in cases where AGC is disabled (i.e. fixed gain control). I have mine set to 0 and the problem disappears.

mjwgeek, 2023-10-30:
I just got home, here is a youtube video link that demonstrates the difference with and without an antenna using the same source.

https://youtube.com/shorts/NhY2-kDljY4?feature=share

There is a 10MB file limit here so this was easier.

OneOfEleven, 2023-10-31:
What's the exact RX frequency the radio is set too on that channel ?

Most all of todays transceivers have internal spurious (interfering) carriers due to various parts of the circuitry, more often than not from the digital parts of the circuit (CPU etc), I'm wondering if you've found one of them (or not).

OneOfEleven, 2023-10-31:
Because that is not AGC based noise, I'm certain it's a sproggy of some kind you've chanced upon, OR, it's being sent as part of the transmission (bad PSU or such like).

mjwgeek, 2023-10-31:
I tried it on 438.600, 146.6100 (in the video) and 467.650, all the same,
but it's not occuring with the latest firmware that was available at the
time of my last post.

mjwgeek, 2023-10-31:
in the video the frequency was tuned to 146.6100.  It's not an issue after I flashed the latest available firmware last night.  And it was happening in 438.600 and 467.650.  I don't believe it's a PSU error as I could flash other firmware and it would go away.  But again, the problem is gone now with the latest firmware that includes MDX RX decoding.  

I can mark this as closed as I flashed the second radio I have and it has the same issue on older firmware too but not on the latest.

closed 2023-10-31 by OneOfEleven