OneOfEleven/uv-k5-firmware-custom #264: Discussion: TransmitSingleTone on SOS?

wrcrooks, 2023-10-29:
So, I noticed a while back when you implemented the Quindar Tone feature that you took it one step further than I would've imagined and instead of just TX'ing the tone, you played it through the audio path/speaker as well for the operator to hear. This blew my mind and when I was messing around with the SOS Flashlight code, I remembered this and was wondering:

Would it be prudent/make any sense to beep out and/or (maybe) transmit that CW 'SOS' as a tone? I don't personally do any hiking so remote that I could get lost but I imagine it could be one of those "better to have it and not need it than need it and not have it" features. 

Opening the floor for a spirited (but still civil) debate on it.

OneOfEleven, 2023-10-29:
Yes no problem, I can get it to tone the SOS in sync with the SOS light, good idea really. I'll do that today sometime.

wrcrooks, 2023-10-29:
You rock! I have some lofty goals with CW that I'd like to implement in this repo if you'll have them. I like the idea of being able to use one of the side buttons to send a call sign in Morse. This would help with FCC identification requirements. I'll create a PR when I have something to show for it.

closed 2023-10-31 by OneOfEleven