OneOfEleven/uv-k5-firmware-custom #265: fw 93c9e59 display layout "NO ISSUE"

IU4FIR, 2023-10-29:
Unfortunately there is no possibility to report suggestions, not issues, however the channel name in SMALL_BOLD is more readable, furthermore the indication of the VFO frequency using TRIM_TRAILING_ZERO is less immediate.  
My personal opinion, just a consideration using glasses -:(

Distinctive features of the fw are lost<<<
ENABLE_SMALL_BOLD  := 0 
ENABLE_RX_SIGNAL_BAR  := 0

Maybe you can recover bytes from these features<<<
ENABLE_TRIM_TRAILING_ZEROS  := 1
ENABLE_SQ_OPEN_WITH_UP_DN_BUTTS  := 1 >>> In my radio not run !

Obviously I can modify it independently, but having no programming experience, for me and many others it is not possible to be autonomous!

Thanks for your attention

[image]

OneOfEleven, 2023-10-29:
Should be OK now to enable small bold and RX bar.

I can honestly say that this firmware's '**distinctive features**' are **_very_** much more than just the small bold font and RX level meter. Easy way to convince yourself is to go back to using the original QS firmware.

IU4FIR, 2023-10-30:
You're absolutely right, I meant only the aesthetic aspect, the rest of the work is great. I apologize if I didn't express myself correctly.

closed 2023-10-30 by IU4FIR

OneOfEleven, 2023-10-30:
Don't worry about it :)

I've had some out there aim there life's frustrations towards me while I've been sharing all this work with them, gets a little deflating at times, so don't be offended, I definitely over react and am a little over defensive at times due to them.

Anyway, enjoy the radio, bugs n all :)  .. I do fix each and every one - eventually, sometimes takes me a while to find them, but I bit by bit I get there.

IU4FIR, 2023-10-30:
You're right and unfortunately people don't understand the life time that
goes into projects.
the improvement that the radio has had with the development of your fw is
fantastic, each of us has our own ideas on the details but unfortunately
not having a compiler simple enough to use (e.g. visual code or other)
would avoid writing observations which are sometimes not so important.
Have a good day!