OneOfEleven/uv-k5-firmware-custom #266: F+* sometimes give a random result when the singal actually have no CTCSS

zhaoparrot, 2023-10-30:
I think i may have seen someone talk about this. not sure.
try to test this in the QS v2.01.31_publish.bin . it did not gives me a wrong result but it comes with the annoying scan fail timeout.
the orange radio is TXing with no CTCSS in this video.

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/27049669/b4c7ceb8-5f1e-4119-9627-bb0dc322154b

I have conducted multiple experiments, and sometimes it works fine, continuously scanning when there is no sub-audible tone in the received signal. However, in rare cases, it provides random analog sub-audible tone results when there is no sub-audible tone in the signal. I suspect this issue won't affect many people, as regular individuals typically don't perform sub-audible tone scanning without the presence of sub-audible tones.

zhaoparrot, 2023-10-30:
this also happens sometime with F+4

zhaoparrot, 2023-10-30:
also, in F+4 mode scaning , the exit button occasionally doesn't respond promptly. Sometimes, it requires multiple presses before it responds.

OneOfEleven, 2023-10-30:
I'll find the shortly, thank you for letting me know.

OneOfEleven, 2023-10-30:
I've updated the code threshold levels and fixed the key pad input, let me know if it's still a bit iffy

zhaoparrot, 2023-10-31:
I conducted some tests, and this feature is working fine now. Thank you so much! I will check your code to see what magic you have performed.

closed 2023-10-31 by zhaoparrot