OneOfEleven/uv-k5-firmware-custom #267: Air copi not work

IT9-TOP, 2023-10-30:
hi, when I add the air function you copy the firmware it doesn't compile it gives me error 1

ENABLE_AIRCOPY                   := 1

OneOfEleven, 2023-10-30:
You're probably running out of flash space, you need to disable some other option(s) if you want to include aircopy, such as MDC1200, UART_DEBUG, FM_RADIO, NOAA, VOICE, VOX, SMALLEST_FONT or SMALL_BOLD etc.

I'm afraid after adding all the various functions people have asked for it's now become impossible to enable many at the same time due to the radios very limited flash program space.

OneOfEleven, 2023-10-30:
Most don't even know what MDC1200 is or means, in which case so disabling that one will help a lot.
None of you need UART_DEBUG so that's another definite.
Make sure SMALLEST_FONT and VOICE are also disabled.

IT9-TOP, 2023-10-30:
I deactivated everything, I only want FM radio and Air Copi, but as soon as I activate Air Copi the firmware doesn't compile, it gives me an error

DualTachyon, 2023-10-30:
2 comments that you say it doesn't compile, but you don't even post the error you get? How do you expect people to help you when you don't provide the details?

IT9-TOP, 2023-10-30:
try to fill it in yourself, it takes 2 seconds, I can't get it to work.

DualTachyon, 2023-10-30:
I did and it works, therefore the onus is on you. You provide no information as to what commit you are using. What OS. What compiler. What errors you get, etc 

You are very rude and I didn't forget your other issue where you insulted me that I thought you were stupid. 

Your comment is invalid because it works for everyone else, so if you don't provide details you're not going to fix your problem.

IT9-TOP, 2023-10-30:
download the firmware package and activate the air copy without touching anything else and you will see that it gives an error, then it is you who is rude to people, we are here to test your work, we are not here to make up stories.

DualTachyon, 2023-10-30:
If you don't post the actual error details, you're not helping us test. You're treating us like slaves. That's now 2 people who confirmed it works. But go ahead and piss off developers with no information whatsoever. A good tester provides a lot of details and does not post "empty" messages.

DualTachyon, 2023-10-30:
[image]

DualTachyon, 2023-10-30:
Torre John, we can see right through you.

IT9-TOP, 2023-10-30:
ENABLE_CLANG                     := 0
ENABLE_SWD                       := 0
ENABLE_OVERLAY                   := 0
ENABLE_LTO                       := 1
# UART Programming 2.9 kB
ENABLE_UART                      := 1
ENABLE_UART_DEBUG                := 0
# AirCopy 2.5 kB
ENABLE_AIRCOPY                   := 1
ENABLE_AIRCOPY_REMEMBER_FREQ     := 1
ENABLE_AIRCOPY_RX_REBOOT         := 0
# FM Radio 4.2 kB
ENABLE_FMRADIO_64_76             := 0
ENABLE_FMRADIO_76_90             := 0
ENABLE_FMRADIO_76_108            := 0
ENABLE_FMRADIO_875_108           := 1
ENABLE_FMRADIO_64_108            := 0
# NOAA 1.2 kB
ENABLE_NOAA                      := 0
# Voice 1.7 kB
ENABLE_VOICE                     := 0
ENABLE_MUTE_RADIO_FOR_VOICE      := 0
# Tx on Voice 1.0 kB
ENABLE_VOX                       := 0
ENABLE_REDUCE_LOW_MID_TX_POWER   := 1
# Tx Alarm 600 B
ENABLE_ALARM                     := 0
ENABLE_TX1750                    := 0
# MDC1200 2.8 kB
ENABLE_MDC1200                   := 1
ENABLE_PWRON_PASSWORD            := 0
ENABLE_RESET_AES_KEY             := 1
ENABLE_BIG_FREQ                  := 0
# smaa bolf 580 B
ENABLE_SMALL_BOLD                := 1
# smallest font 2 kB
ENABLE_SMALLEST_FONT             := 0
# trim trailing 44 B
ENABLE_TRIM_TRAILING_ZEROS       := 1
ENABLE_KEEP_MEM_NAME             := 1
ENABLE_WIDE_RX                   := 1
ENABLE_TX_WHEN_AM                := 0
# Freq calibration 188 B
ENABLE_F_CAL_MENU                := 0
ENABLE_TX_UNLOCK                 := 0
ENABLE_CTCSS_TAIL_PHASE_SHIFT    := 1
ENABLE_CONTRAST                  := 0
ENABLE_BOOT_BEEPS                := 0
ENABLE_DTMF_CALL_FLASH_LIGHT     := 1
ENABLE_FLASH_LIGHT_SOS_TONE      := 1
ENABLE_SHOW_CHARGE_LEVEL         := 0
ENABLE_REVERSE_BAT_SYMBOL        := 1
ENABLE_FREQ_SEARCH_TIMEOUT       := 1
ENABLE_CODE_SEARCH_TIMEOUT       := 1
# Kill and Revive 400 B
ENABLE_KILL_REVIVE               := 0
# AM Fix 800 B
ENABLE_AM_FIX                    := 1
ENABLE_AM_FIX_SHOW_DATA          := 0
# Squelch 12 B .. can't be right ?
ENABLE_SQUELCH_MORE_SENSITIVE    := 1
ENABLE_SQ_OPEN_WITH_UP_DN_BUTTS  := 1
ENABLE_FASTER_CHANNEL_SCAN       := 1
ENABLE_COPY_CHAN_TO_VFO_TO_CHAN  := 1
# Rx Signal Bar 400 B
ENABLE_RX_SIGNAL_BAR             := 1
# Tx Timeout Bar 200 B
ENABLE_TX_TIMEOUT_BAR            := 0
# Tx Audio Bar 300 B
ENABLE_TX_AUDIO_BAR              := 1
# Side Button Menu 300 B
ENABLE_SIDE_BUTT_MENU            := 1
# Key Lock 400 B
ENABLE_KEYLOCK                   := 0
#ENABLE_PANADAPTER               := 0
#ENABLE_SINGLE_VFO_CHAN          := 0

IT9-TOP, 2023-10-30:
[image]

DualTachyon, 2023-10-30:
Was that too much to ask? Nex time provide all the details from the beginning and you will have more help.

You enabled too many features, you need to disable some since they don't fit.

DualTachyon, 2023-10-30:
You said you only want Aircopy and FM, so disable MDC1200 and it will fit.

DualTachyon, 2023-10-30:
OneOfEleven told you in the very first reply you have to disable something. Don't complain you didn't get help or a solution.

IT9-TOP, 2023-10-30:
ok thanks

closed 2023-10-30 by IT9-TOP

DualTachyon, 2023-10-30:
This is probably what your problem was: after you change ENABLE_xxx, you must ALWAYS "make clean" before "make".