OneOfEleven/uv-k5-firmware-custom #270: Problem in reception of some stations

Zahurzahur, 2023-10-30:
During reception of a station which was connecting with many other stations. My K5 radio was receving comunitions with cuts even  signals were strong. With monitor on there were no cuts. I m testing again and again. This isue is with few stations. My Kenwood has no any isue like that .  I have tried old FWs but isue was present even  in stock FW. 
Now I m testing on other stations, even on strongest signal the RSS bar is also fluctuating. This is happening with particular stations.

OneOfEleven, 2023-10-30:
Need to see and hear a short video of it happening, ignore the RSSI bar for now, that's purely cosmetic.

rai4424, 2023-10-30:
How can I post video?

Zahurzahur, 2023-10-31:
https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/149041689/792c7448-fa89-4a28-b462-15d6c2de12bf

Zahurzahur, 2023-10-31:
Above video is with squel on- squelch off- squelch on.

OneOfEleven, 2023-10-31:
ah, set the channel to **WIDE** mode, the signal looks to be too wide for narrow mode (you have that channel set to **NARROW** mode).

Let me know how it goes.

Zahurzahur, 2023-10-31:
I m so 😜

closed 2023-10-31 by OneOfEleven