OneOfEleven/uv-k5-firmware-custom #272: Monitor button does not disable monitor on second press

tevevizor, 2023-10-30:
As the title says, when monitor is enabled, pressing the button again will not disable it (enable the squelch again).
Only once I press the PTT, monitor shuts off and squelch is back to working as normal.

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/14348388/0023cca9-b666-43ab-950f-27953bbab4de

tevevizor, 2023-10-30:
This is my makefile:

`ENABLE_CLANG                     := 0
ENABLE_SWD                       := 0
ENABLE_OVERLAY                   := 0
ENABLE_LTO                       := 1
# UART Programming 2.9 kB
ENABLE_UART                      := 1
ENABLE_UART_DEBUG                := 0
# AirCopy 2.5 kB
ENABLE_AIRCOPY                   := 0
ENABLE_AIRCOPY_REMEMBER_FREQ     := 0
ENABLE_AIRCOPY_RX_REBOOT         := 0
# FM Radio 4.2 kB
ENABLE_FMRADIO_64_76             := 0
ENABLE_FMRADIO_76_90             := 0
ENABLE_FMRADIO_76_108            := 0
ENABLE_FMRADIO_875_108           := 0
ENABLE_FMRADIO_64_108            := 0
# NOAA 1.2 kB
ENABLE_NOAA                      := 0
# Voice 1.7 kB
ENABLE_VOICE                     := 0
ENABLE_MUTE_RADIO_FOR_VOICE      := 0
# Tx on Voice 1.0 kB
ENABLE_VOX                       := 0
ENABLE_REDUCE_LOW_MID_TX_POWER   := 1
# Tx Alarm 600 B
ENABLE_ALARM                     := 1
ENABLE_TX1750                    := 1
# MDC1200 2.8 kB
ENABLE_MDC1200                   := 0
ENABLE_PWRON_PASSWORD            := 0
ENABLE_RESET_AES_KEY             := 1
ENABLE_BIG_FREQ                  := 1
# smaa bolf 580 B
ENABLE_SMALL_BOLD                := 1
# smallest font 2 kB
ENABLE_SMALLEST_FONT             := 0
# trim trailing 44 B
ENABLE_TRIM_TRAILING_ZEROS       := 0
ENABLE_KEEP_MEM_NAME             := 1
ENABLE_WIDE_RX                   := 1
ENABLE_TX_WHEN_AM                := 0
# Freq calibration 188 B
ENABLE_F_CAL_MENU                := 0
ENABLE_TX_UNLOCK                 := 0
ENABLE_CTCSS_TAIL_PHASE_SHIFT    := 1
ENABLE_CONTRAST                  := 1
ENABLE_BOOT_BEEPS                := 0
ENABLE_DTMF_CALL_FLASH_LIGHT     := 1
ENABLE_FLASH_LIGHT_SOS_TONE      := 1
ENABLE_SHOW_CHARGE_LEVEL         := 1
ENABLE_REVERSE_BAT_SYMBOL        := 1
ENABLE_FREQ_SEARCH_TIMEOUT       := 1
ENABLE_CODE_SEARCH_TIMEOUT       := 1
# Kill and Revive 400 B
ENABLE_KILL_REVIVE               := 0
# AM Fix 800 B
ENABLE_AM_FIX                    := 1
ENABLE_AM_FIX_SHOW_DATA          := 0
# Squelch 12 B .. can't be right ?
ENABLE_SQUELCH_MORE_SENSITIVE    := 0
ENABLE_SQ_OPEN_WITH_UP_DN_BUTTS  := 0
ENABLE_FASTER_CHANNEL_SCAN       := 1
ENABLE_COPY_CHAN_TO_VFO_TO_CHAN  := 1
# Rx Signal Bar 400 B
ENABLE_RX_SIGNAL_BAR             := 1
# Tx Timeout Bar 200 B
ENABLE_TX_TIMEOUT_BAR            := 0
# Tx Audio Bar 300 B
ENABLE_TX_AUDIO_BAR              := 1
# Side Button Menu 300 B
ENABLE_SIDE_BUTT_MENU            := 1
# Key Lock 400 B
ENABLE_KEYLOCK                   := 1
#ENABLE_PANADAPTER               := 0
#ENABLE_SINGLE_VFO_CHAN          := 0`

Zahurzahur, 2023-10-31:
Monitor behavior was odd just after installing newest bin file.  Using SQUELCH option harder then softer, now its ok.  Some time it behave like that.

tevevizor, 2023-10-31:
> Monitor behavior was odd just after installing newest bin file. Using SQUELCH option and set it on 2 ( Squelch 01 was turning monitor on), now its ok. I have also set side button1 for monitor only (short and long press).

Setting SQL to 2 doesn't resolve the monitor issue for me, although I noticed when higher squelch sensitivity is enabled, any noise breaks the squelch.
Also, both short and long press on button 1 is selected as monitor button on mine also.

closed 2023-10-31 by tevevizor