OneOfEleven/uv-k5-firmware-custom #273: RX_SIGNAL_BAR is gone

zhaoparrot, 2023-10-31:
I have MDC1200 enabled and the rx bar is gone, thinking it may be related to
```
#ifdef ENABLE_MDC1200
			if (mdc1200_rx_ready_tick_500ms > 0)
			{
				g_center_line = CENTER_LINE_MDC1200;
```
so i try disable MDC1200
RXbar still no show
my makefile ↓
```
ENABLE_CLANG                     := 0
ENABLE_SWD                       := 0
ENABLE_OVERLAY                   := 0
ENABLE_LTO                       := 1
# UART Programming 2.9 kB
ENABLE_UART                      := 1
ENABLE_UART_DEBUG                := 0
# AirCopy 2.5 kB
ENABLE_AIRCOPY                   := 0
ENABLE_AIRCOPY_REMEMBER_FREQ     := 0
ENABLE_AIRCOPY_RX_REBOOT         := 0
# FM Radio 4.2 kB
ENABLE_FMRADIO_64_76             := 0
ENABLE_FMRADIO_76_90             := 0
ENABLE_FMRADIO_76_108            := 0
ENABLE_FMRADIO_875_108           := 0
ENABLE_FMRADIO_64_108            := 0
# NOAA 1.2 kB
ENABLE_NOAA                      := 0
# Voice 1.7 kB
ENABLE_VOICE                     := 0
ENABLE_MUTE_RADIO_FOR_VOICE      := 0
# Tx on Voice 1.0 kB
ENABLE_VOX                       := 0
ENABLE_REDUCE_LOW_MID_TX_POWER   := 1
# Tx Alarm 600 B
ENABLE_ALARM                     := 0
ENABLE_TX1750                    := 0
# MDC1200 2.8 kB
ENABLE_MDC1200                   := 0
ENABLE_PWRON_PASSWORD            := 0
ENABLE_RESET_AES_KEY             := 0
ENABLE_BIG_FREQ                  := 1
# smaa bolf 580 B
ENABLE_SMALL_BOLD                := 1
# smallest font 2 kB
ENABLE_SMALLEST_FONT             := 1
# trim trailing 44 B
ENABLE_TRIM_TRAILING_ZEROS       := 0
ENABLE_KEEP_MEM_NAME             := 1
ENABLE_WIDE_RX                   := 0
ENABLE_TX_WHEN_AM                := 0
# Freq calibration 188 B
ENABLE_F_CAL_MENU                := 0
ENABLE_TX_UNLOCK                 := 0
ENABLE_CTCSS_TAIL_PHASE_SHIFT    := 0
ENABLE_CONTRAST                  := 0
ENABLE_BOOT_BEEPS                := 0
ENABLE_DTMF_CALL_FLASH_LIGHT     := 0
ENABLE_FLASH_LIGHT_SOS_TONE      := 0
ENABLE_SHOW_CHARGE_LEVEL         := 0
ENABLE_REVERSE_BAT_SYMBOL        := 0
ENABLE_FREQ_SEARCH_TIMEOUT       := 0
ENABLE_CODE_SEARCH_TIMEOUT       := 0
# Kill and Revive 400 B
ENABLE_KILL_REVIVE               := 0
# AM Fix 800 B
ENABLE_AM_FIX                    := 1
ENABLE_AM_FIX_SHOW_DATA          := 0
# Squelch 12 B .. can't be right ?
ENABLE_SQUELCH_MORE_SENSITIVE    := 1
ENABLE_SQ_OPEN_WITH_UP_DN_BUTTS  := 1
ENABLE_FASTER_CHANNEL_SCAN       := 1
ENABLE_COPY_CHAN_TO_VFO_TO_CHAN  := 1
# Rx Signal Bar 400 B
ENABLE_RX_SIGNAL_BAR             := 1
# Tx Timeout Bar 200 B
ENABLE_TX_TIMEOUT_BAR            := 0
# Tx Audio Bar 300 B
ENABLE_TX_AUDIO_BAR              := 0
# Side Button Menu 300 B
ENABLE_SIDE_BUTT_MENU            := 0
# Key Lock 400 B
ENABLE_KEYLOCK                   := 1
#ENABLE_PANADAPTER               := 0
#ENABLE_SINGLE_VFO_CHAN          := 0
```
by the way MDC1200 does not take 2.8 kB flash in my test, more like 2.3kB to me .

OneOfEleven, 2023-10-31:
There are menu options to enable/disable the TX and TX bars, somewhere around menu number 33 or so. check to make sure the RX bar menu is set to **ON**.

zhaoparrot, 2023-10-31:
Ah!  it's a menu option now , good to know . works fine then. thanks.

closed 2023-10-31 by zhaoparrot