OneOfEleven/uv-k5-firmware-custom #275: Broadcast FM reception

IU4FIR, 2023-10-31:
**I'm sorry to add another report :'-( , this one concerns the behavior of FM reception, I try to explain in the attached image..**
[image]

IU4FIR, 2023-10-31:
As I don't use FM very often, am I using the wrong commands?

OneOfEleven, 2023-10-31:
It will ONLY go into FM radio mode if it's NOT receiving a signal, it will also come out of FM radio mode if it later receives a signal - the radio's main purpose is to be a 2-way radio (walkie talkie), the FM radio is an add-on that always takes 2nd place.

use the exit button to exit anything, including FM

As this radio is destined for the radio amateur market (it's advertised and sold as such), the user is meant to have a radio transmitting license and so therefore an understanding/technical knowledge of radio communications and how it works.
I can't keep having to help with issues where the user just doesn't know what they are doing anymore. If others want to step in and do that then they are totally welcome.

closed 2023-10-31 by OneOfEleven