OneOfEleven/uv-k5-firmware-custom #276: closing github account

OneOfEleven, 2023-10-31:
I'm closing down this github account soon, so if you want a copy of this firmware please make sure you grab it asap.

github are telling me my access to github is going to be somewhat limited due to not handing over my phone details or installing more software onto the PC. I can't do either so grab while you can !

Tunas1337, 2023-10-31:
Hi 1of11, I'm very sorry to hear that... GitHub isn't quite the same since Micro$oft got their grubby hands on it.

Would you consider rehosting it on Gitlab or a similar less-restrictive platform, or is this the end of the line? Thank you so much for your work, either way.

DualTachyon, 2023-10-31:
Let me know if my latest 2FA advice worked for you.

ea7hjb, 2023-10-31:
Hope to see you around. Thanks for your time and be well!

mjwgeek, 2023-11-01:
Oh wow, well you worked hard and made some cool additions to this cool radio.  Thank you and everyone who contributed.

chaolucky18, 2023-11-01:
hope to see you again, i really love your project~
maybe you can transfer to gitlab or another platform?

TurnOffNOD, 2023-11-01:
Are there any details about github's warning?

OneOfEleven, 2023-11-01:
It's bad enough that google has my phone number (android OS on the phone), but microsoft aren't having it, too much data mining going on now.

I'm trying a little old windows program out that DualTachyon kindly pointed me too, seems to be OK - for now, If it stops microsoft demanding my number I'll stay with github, we'll see.

ANTodorov, 2023-11-01:
First, I want to thank you for the great work!
I'll honour any decision you make, but I hope this issue to be resolved.  There are indeed less demanding alternatives than GitHub.

motorello, 2023-11-01:
OneOfEleven, I'm not used to post on the internet, but this time I also want to thank you for your amazing effort to transform a crappy radio in an pleasant tool that gave me the chance to enjoy a whole afternoon listening to atc communications.
In addition to this, I also forked your repo, learnt how to use git, made some mods in the ctcss scan and now I could be ready to submit my first PR. As I wrote on TG, I'm 50, but in the last few months I learnt a lot, just like a child who is discovering the world.
It's a pity that this amazing work get ruined by some folks...
Thank you again,
All the best!

ea7hjb, 2023-11-01:
> It's bad enough that google has my phone number (android OS on the phone), but microsoft aren't having it, too much data mining going on now.

Sad part of this is that, most likely, Microsoft already know your phone number. The data mining, as you say, is brutal nowadays. Only partial fix I have found for this privacyless life we are to live is to have two phone numbers: one for my family and friends and important things, and other for the rest of bullshit, including work. Because if your boss has your private phone it's not better than Microsoft.

closed 2023-11-02 by OneOfEleven