OneOfEleven/uv-k5-firmware-custom #279: Dual Watch not working

dutn, 2023-11-01:
First of all, thank you for your great work!

starting 2 day's ago (30-10-2023); the firmware doesn't allow me to listen to both frequency at the same time. DW symbol is on, but it only receives on the frequency that is selected. Did try to reflash, change DW mode; but all the same.

I don't see it reported here, but if it is a mistake by myself; please close. Thank you.

krotoniec, 2023-11-01:
I think there is something strange going on with DW indeed. Right now I observe that after transmission on one channel when DW indicators is back already, I'm unable to receive on the other one right away, it takes some additional several seconds before I'm able to receive the transmission.

ea7hjb, 2023-11-01:
I tested the latest compilation and I think the DW is working as expected. Note that this NOT a true Dual VFO device (like a VX-7 was, for example). What this device does is quickly switch between the two set frequencies and stop on the one where it detects activity. The receiver will stay on that VFO while there is a transmission, and will remain there for and extra 10 seconds (aprox.) more after the transmission has ceased, and then it will resume the dual monitorization again.

This is why if you have DW activated and you receive a TX in a VFO the other VFO will be useless for a while.

dutn, 2023-11-01:
Thank you @ea7hjb for your answer.

To be honest, it doesn't sound the same to me. If I select channel B (that is not receiving), then channel A is not receiving, even if there is a signal. If I switch to channel A (select channel A), then it starts receiving on that channel.

It is not a waiting time. Before the update on the 30th, Dual watch was working as expected (UV-K5).

Hope, it helps.
Peter.

OneOfEleven, 2023-11-02:
I'll have a look and see how it goes. I've made some huge changes in the past day or two, so I expect some fall out from that, the odd bug will show itself, I just need feedback on ANY that anyone may find or notice etc.

OneOfEleven, 2023-11-02:
Do you have RX CTCSS or CDCSS enabled on either vfo at all ? .. I just need to set the same conditions so I can reproduce the problem to find the bug

OneOfEleven, 2023-11-02:
Does the problem go away if you disable power save ?  .. power save/battery save is the BatSAV menu setting, set it to OFF, let me know if the bug goes away.

dutn, 2023-11-02:
Thank you @OneOfEleven;

Tested with CTCSS (TX and RX) on and off, for both channels and all combinations; but doesn't make any difference. Both channels do the same, it will not receive the other channel (with DW enabled).

PowerSave was off, but with off or enabled 1:2 or higher, is the same.

It looks exactly if DW is not enabled (but it is and the icon is displayed). Tested with UHF and VHF.
Again, thank you for your great work. I'm using your firmware as the daily driver (even with this strange DW issue ;-) ).

OneOfEleven, 2023-11-02:
Try it now, let me know how it goes.

dutn, 2023-11-02:
@OneOfEleven ; thank you for the fast action. I'll check today and let you know.

mjwgeek, 2023-11-02:
I tested it and it's working much faster, ie, it's detecting transmissions on a/b within a second or less, at least for me.  At first it through me for a loop then I noticed an new option besides FM or AM, DSB.  Hopefully SSB soon :)

OneOfEleven, 2023-11-02:
mmmm I added DSB a few whiles ago, you're the first to mention it.
I'm afraid DSB is as good as it gets, I know fagci calls it SSB, but it's actually DSB, their are two choices, both are DSB, so I called it DSB rather than LSB/USB/SSB. I've yet to receive any SSB using it to test it.

The dual watch togglse between the two VFO frequencies/channels once every 100ms, if it receives a signal it pauses for however long your have '**S HOLD**' set too (menu option) after the signal vanishes, at which t continues to toggle again.

mjwgeek, 2023-11-02:
I only noticed it because for some reason it changed from FM to DSB on it's own, another quirky bug I suppose.  Cool to have it as a reception options.  Good to know about the S Hold options, I'll check it out!

OneOfEleven, 2023-11-02:
It changed on it's own ?? .. that should NOT be possible.

You can toggle the AM/FM/DSB mode by pressing F+0 key, or by changing it in the menu, or by a channel change, but that's it.

What were you doing when the radios mode changed by itself ? .. that's a bug that MUST be caught.

mjwgeek, 2023-11-02:
Oh, F+0 changes it, which I've not noticed before because I've always had FM mode enabled which I had to disable to flash the latest build to save space.  So, let's call this one user induced :)

OneOfEleven, 2023-11-02:
ah !

To turn the radio just do a long press on the 0/FM button. If the F-key is used the 0/FM button toggles the modulation mode.

closed 2023-11-02 by OneOfEleven

mjwgeek, 2023-11-02:
Clever!

dutn, 2023-11-02:
Great work @OneOfEleven ; Dual Watch is working again. Thank you for the quick action.