OneOfEleven/uv-k5-firmware-custom #284: When saving a channel, keeping name the name, it combines two different channel names.  Appears after Restart

mjwgeek, 2023-11-02:
I've confirmed this multiple times now.

I have on channel 23 the name AllstarV.  I have on Channel 32 the name Fort 550.  After saving channel 23 from the radio, then turning off the radio and turning it back on, Channel 23 now says AllstarV Fort 550.  

I tested with Channel 22 with the name AllstarU and it added the name of my Channel 31 to the end of Channel 22 after a restart.  

So it seems to be pulling information from 8 channels down.

Build Version bb3b681

Just reporting bugs as you asked.

OneOfEleven, 2023-11-02:
Think I've fixed it, let me know if not.

mjwgeek, 2023-11-02:
I can't get past this when building:

functions.c: In function 'FUNCTION_Select':
functions.c:268:22: error: 'vfo_info_t' has no member named 'mdc1200_mode'
  268 |     if (g_current_vfo->mdc1200_mode == MDC1200_MODE_BOT || g_current_vfo->mdc1200_mode == MDC1200_MODE_BOTH)
      |                      ^~
functions.c:268:73: error: 'vfo_info_t' has no member named 'mdc1200_mode'
  268 |     if (g_current_vfo->mdc1200_mode == MDC1200_MODE_BOT || g_current_vfo->mdc1200_mode == MDC1200_MODE_BOTH)
      |                                                                         ^~
make: *** [functions.o] Error 1

mjwgeek, 2023-11-03:
Now with the latest version, after flashing, the screen is blank.  Tx still works, but there is nothing displayed on the screen.

closed 2023-11-03 by OneOfEleven