OneOfEleven/uv-k5-firmware-custom #287: MDC1200 compilation error for 8dfcbed

ETO-bigbang, 2023-11-02:
I used 8dfcbed to compile MDC1200 and reported an error

I enabled the MDC1200 function in MAKEFILE

```
functions.c: In function 'FUNCTION_Select':
functions.c:268:22: error: 'vfo_info_t' has no member named 'mdc1200_mode'
  268 |     if (g_current_vfo->mdc1200_mode == MDC1200_MODE_BOT || g_current_vfo->mdc1200_mode == MDC1200_MODE_BOTH)
      |                      ^~
functions.c:268:73: error: 'vfo_info_t' has no member named 'mdc1200_mode'
  268 |     if (g_current_vfo->mdc1200_mode == MDC1200_MODE_BOT || g_current_vfo->mdc1200_mode == MDC1200_MODE_BOTH)
      |                                                                         ^~
make: *** [functions.o] 错误 1
```

OneOfEleven, 2023-11-03:
Thank you for letting me know.

What's the 'power section' ? .. do you mean TX power ?

What's the 'K5' button ?

ETO-bigbang, 2023-11-03:
TX power Unable to save.

M,UP,DOWN,EXIT,numeral key，all key.

and ENABLE_BOOT_BEEPS := 1, open the radio,other radio RX beep.

OneOfEleven, 2023-11-03:
The TX power is saved. It's a channel setting. When ever a little 'ch' appears in the sub-menu area, it means that particular setting is a per channel setting, that means after changing it you have to re-save the channel for it to be remembered.

ETO-bigbang, 2023-11-03:
> The TX power is saved. It's a channel setting. When ever a little 'ch' appears in the sub-menu area, it means that particular setting is a per channel setting, that means after changing it you have to re-save the channel for it to be remembered.

This is a great reminder.
TX power cannot be selected in the menu. When you select Low and press M to save, the result remains High and does not change to LOW. Then, save the channel again

ETO-bigbang, 2023-11-03:
I tested 7E5605C and fixed 1 and 2. Thank you very much for your perfect repair.

ETO-bigbang, 2023-11-03:
I tested 7E5605C and you have fixed 1 and 2. Thank you very much for your perfect repair.

ETO-bigbang, 2023-11-03:
I tested 7E5605C and you have fixed it. Thank you very much for the perfect repair.

closed 2023-11-03 by ETO-bigbang