OneOfEleven/uv-k5-firmware-custom #288: The FM radio became silent

ETO-bigbang, 2023-11-03:
I noticed that after the keyboard sound emission issue was resolved, the FM radio became silent.

OneOfEleven, 2023-11-03:
I disabled the FM radio to test and fix an MDC1200 compile bug, I'm afraid you can't have all the options enabled at once, not enough flash area anymore. If you want the FM radio you need to select the compile options you want and run the compile to create the firmware you want.

ETO-bigbang, 2023-11-03:
> I disabled the FM radio to test and fix an MDC1200 compile bug, I'm afraid you can't have all the options enabled at once, not enough flash area anymore. If you want the FM radio you need to select the compile options you want and run the compile to create the firmware you want.

I chose the FM function for compilation and found that there was no sound in FM. I looked at your makefile file.

dutn, 2023-11-03:
For me; FM radio works when Dual Watch is disabled (with DW enabled, it is silent).

ETO-bigbang, 2023-11-03:
> For me; FM radio works when Dual Watch is disabled (with DW enabled, it is silent).

me too

OneOfEleven, 2023-11-03:
okey doke, was working on something else for a while then. I'll go find the problem

You might find other bugs too because I made a **HUGE** change to the code past few days, I now load the entire EEPROM contents into memory at boot up in order to drastically speed things up from that point on.

So keep creating the issues as you find any bugs - _**a separate issue for each bug please**_, otherwise any added-on bug reports will get deleted when I close the original issue reason, and so not get delt with (my memory is not so long term these days).

ETO-bigbang, 2023-11-03:
> okey doke, was working on something else for a while then. I'll go find the problem
>
> You might find other bugs too because I made a **HUGE** change to the code past few days, I now load the entire [...]

Yes, I noticed that the response speed of k5 has increased. 
I saw that you have made significant changes, and I know you are busy with other issues. Next time, I will create new issues. 
You have been working too hard lately and need to take a break.

closed 2023-11-04 by ETO-bigbang

ETO-bigbang, 2023-11-04:
Fixed. Thank you sir!