OneOfEleven/uv-k5-firmware-custom #297: Round amount of frequency/ step adjustment

SteefW, 2023-11-04:
You have done so much. And all devs. It is genuinely incredible. Thanks so much! However maybe there is one small thing that might be possible without much extra space? How about respecting starting freq on sepsize. In my AOR5000 it is called stepadjustment i think (not comparing devices just the function). So eg on 11m band stepsize 10kc is ‘starting’at a certain freq 26.965. Step size 12,5kc starting at 446.00625 at not 446.000. Now the freq changes automatically to a ‘round’ number. I think this is might be a bug also? I know I am weening because so much important work is already done, but on the other hand it would make operating vfo so much more easy with also a lot advantages. Searching will be quicker and more right, Spec will be wider as well etc…. Thanks for reading the idea… / wish 😁

OneOfEleven, 2023-11-04:
I've done that, you need to enter/start from a  spot frequency with the desired offset (ie, + 1.25kHz, 6.25kHz etc) , the up/dn/scan will then go in the set frequency step size but retains the initial offset.

closed 2023-11-04 by OneOfEleven

SteefW, 2023-11-04:
Hi I use fw vd59d4ea from this morning. 
I do not unfortunately what you mean. I put in the right freq et 446.00625 in vfo mode. This only work if the step size is set to 6,25kc.
Then I would like to set step size to 12,5kc. Because next channel is 12,5kc away. But when changing the step size to 12,5kc the fw will then automatically change freq as well. It will round up or down and leave the desired freq of 446.00625. If there ia already a solution for this in vfo mode I hope you want to explain me again. I  sorry… thanks so much !

OneOfEleven, 2023-11-04:
Sounds like you're using old firmware ?

SteefW, 2023-11-04:
Really?
I started my last message with:
“ Hi I use fw : d59d4ea from this morning! Is this too old?

OneOfEleven, 2023-11-04:
yes, too old

SteefW, 2023-11-04:
Ok. So it is recently changed. Great! Thanks so much! Which fw should I use? For it seems not yet be posted in the telegram fw group?

OneOfEleven, 2023-11-04:
???

SteefW, 2023-11-04:
Succeeded. See you changed it around 10h this morning. Sounded like it was weeks ago heheh. Thanks so much man for the trouble !! Greets Steef