OneOfEleven/uv-k5-firmware-custom #298: cannot build latest, something problem in `audio.c`

493505110, 2023-11-04:
```
zhou2008@zhou2008-Laptop:/mnt/d/Desktop/uv-k5-firmware-custom$ make
GIT_HASH = 51ac6fc
TOP = /mnt/d/Desktop/uv-k5-firmware-custom
arm-none-eabi-gcc -Os -Werror -mcpu=cortex-m0 -freorder-blocks-algorithm=stc -std=c11 -MMD -flto=2 -Wall -Wextra -Wpedantic -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"51ac6fc\" -DENABLE_FMRADIO_64_108 -DENABLE_FMRADIO -DENABLE_UART -DENABLE_BIG_FREQ -DENABLE_VOICE -DENABLE_ALARM -DENABLE_TX1750 -DENABLE_MDC1200 -DENABLE_MDC1200_SHOW_OP_ARG -DENABLE_KEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_CONTRAST -DENABLE_DTMF_CALL_FLASH_LIGHT -DENABLE_FLASH_LIGHT_SOS_TONE -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_SCAN_IGNORE_LIST -DENABLE_AM_FIX -DENABLE_SQUELCH_MORE_SENSITIVE -DENABLE_SQ_OPEN_WITH_UP_DN_BUTTS -DENABLE_FASTER_CHANNEL_SCAN -DENABLE_RX_SIGNAL_BAR -DENABLE_COPY_CHAN_TO_VFO_TO_CHAN -DENABLE_SIDE_BUTT_MENU -I /mnt/d/Desktop/uv-k5-firmware-custom -I /mnt/d/Desktop/uv-k5-firmware-custom/external/CMSIS_5/CMSIS/Core/Include/ -I /mnt/d/Desktop/uv-k5-firmware-custom/external/CMSIS_5/Device/ARM/ARMCM0/Include -c audio.c -o audio.o
audio.c: In function 'AUDIO_PlaySingleVoice':
audio.c:341:24: error: 'g_rx_vfo' undeclared (first use in this function); did you mean 'g_rx_vfo_num'?
  341 |     AUDIO_set_mod_mode(g_rx_vfo->am_mode);
      |                        ^~~~~~~~
      |                        g_rx_vfo_num
audio.c:341:24: note: each undeclared identifier is reported only once for each function it appears in
audio.c: In function 'AUDIO_PlayQueuedVoice':
audio.c:489:23: error: 'g_rx_vfo' undeclared (first use in this function); did you mean 'g_rx_vfo_num'?
  489 |    AUDIO_set_mod_mode(g_rx_vfo->am_mode);
      |                       ^~~~~~~~
      |                       g_rx_vfo_num
make: *** [Makefile:528: audio.o] Error 1
zhou2008@zhou2008-Laptop:/mnt/d/Desktop/uv-k5-firmware-custom$
```

493505110, 2023-11-04:
and i replace it with `g_rx_vfo_num`, i got this error
```
audio.c: In function 'AUDIO_PlaySingleVoice':
audio.c:341:36: error: invalid type argument of '->' (have 'int')
  341 |     AUDIO_set_mod_mode(g_rx_vfo_num->am_mode);
      |                                    ^~
audio.c: In function 'AUDIO_PlayQueuedVoice':
audio.c:489:35: error: invalid type argument of '->' (have 'int')
  489 |    AUDIO_set_mod_mode(g_rx_vfo_num->am_mode);
      |     
```

OneOfEleven, 2023-11-04:
fixed.

You must be the first to enable voice.

closed 2023-11-04 by OneOfEleven