OneOfEleven/uv-k5-firmware-custom #300: Incorrect frequency reading on the 11m band

Drevenezelezo, 2023-11-04:
Hello.  When displayed in the EU CB band on the main screen, the frequency 108Mhz is displayed on all channels.  However, the frequency is correct in the 20.56 menu.  The last version where it worked correctly was from 19.10. I don't understand programming at all, please don't take it as criticism :) This is just an observation of an ordinary user who appreciates your work.

[image]
[image]

OneOfEleven, 2023-11-04:
Looks like a problem with your windows programming software, I've seen it before with the QS software and chirp. They don't recognise 27MHz as being a valid frequency and so mess up.

Your only choice is to edit the channels on the radio itself - tedious and long winded but not much choice.

I can maybe add some code in the firmware to check each channel at boot-up to ensure the band and channel match up.

OneOfEleven, 2023-11-04:
Try the firmware now, you'll need to re-upload your channels from the PC after uploading the new firmware.

Drevenezelezo, 2023-11-04:
After uploading the firmware, the frequencies are displayed correctly.  Thank you 🙂

closed 2023-11-04 by OneOfEleven