OneOfEleven/uv-k5-firmware-custom #301: UP/DOWN buttons are changing frequency on both channel at once

krotoniec, 2023-11-04:
Great work as usual, feature packed firmware that brings a totally new life to these little radios. 
Today I noticed very weird behavior though. While in frequency mode on both channels, up/down buttons are changing frequency on both channels, not only on the active one. 
Steps to reproduce:
1. 22c7754 flashed.
2. DW mode active
3. Both channels in frequency mode set to 145Mhz, 25KHz step
4. Say, top channel active, bottom one will do just fine.
5. Press up several times, then down several times and observe frequency on both channels.

OneOfEleven, 2023-11-04:
Try it now, let me know if it's fixed.

krotoniec, 2023-11-04:
Th frequency change appears to be fixed on 974538a, but now I'm unable to type in frequency manually while in VFO. Only up/down change works.
I forgot to mention that earlier, that was also the problem with fca972a.

OneOfEleven, 2023-11-05:
I can type the frequency in, but I broke the VFO band, which should be OK now.

OneOfEleven, 2023-11-05:
Should be OK now

closed 2023-11-05 by OneOfEleven