OneOfEleven/uv-k5-firmware-custom #302: contrast collaps when down key is pressed continuously

pawol, 2023-11-04:
On commit 29a77d1 contrast of lcd degrades when M key pressed and aftert that down key pressed continuosly

OneOfEleven, 2023-11-06:
I can't make mine do that.

pawol, 2023-11-06:
I did not use egzumer since 1 month. I use your bilds every day

pawol, 2023-11-06:
Solution with egzumer build helped

OneOfEleven, 2023-11-06:
I have no idea what was wrong or anything, but it sounds like it's all to do with a another firmware ?

Unless there is something I can change in this firmware to stop it happening again ?

OneOfEleven, 2023-11-06:
> If you are using my compiled builds then that suggest you have issues with you radio. I Have only ever had the issue you described when moving between egzumers.. and one of my 7 radios is always using my 1of11 build as its my wifes preferred FW and no such issues like yours..

ah ha !