OneOfEleven/uv-k5-firmware-custom #303: Direction Bug error while building latest

mjwgeek, 2023-11-05:
app/main.c: In function 'MAIN_Key_UP_DOWN':
app/main.c:1105:162: error: 'Direction' undeclared (first use in this function); did you mean 'direction'?
 1105 |     Channel = NOAA_CHANNEL_FIRST + NUMBER_AddWithWraparound(g_eeprom.config.setting.indices.vfo[g_eeprom.config.setting.tx_vfo_num].screen - NOAA_CHANNEL_FIRST, Direction, 0, 9);
      |
                                                 ^~~~~~~~~
      |
                                                 direction
app/main.c:1105:162: note: each undeclared identifier is reported only once for each function it appears in
make: *** [app/main.o] Error 1

ENABLE_CLANG                     := 0
ENABLE_SWD                       := 0
ENABLE_OVERLAY                   := 0
ENABLE_LTO                       := 1
# UART Programming 2.9 kB
ENABLE_UART                      := 1
ENABLE_UART_DEBUG                := 0
# AirCopy 2.5 kB
ENABLE_AIRCOPY                   := 0
ENABLE_AIRCOPY_REMEMBER_FREQ     := 1
ENABLE_AIRCOPY_RX_REBOOT         := 0
# FM Radio 4.2 kB
ENABLE_FMRADIO_64_76             := 0
ENABLE_FMRADIO_76_90             := 0
ENABLE_FMRADIO_76_108            := 0
ENABLE_FMRADIO_875_108           := 0
ENABLE_FMRADIO_64_108            := 1
# NOAA 1.2 kB
ENABLE_NOAA                      := 1
# Voice 1.7 kB
ENABLE_VOICE                     := 0
ENABLE_MUTE_RADIO_FOR_VOICE      := 0
# Tx on Voice 1.0 kB
ENABLE_VOX                       := 0
ENABLE_VOX_MORE_SENSITIVE        := 0
ENABLE_REDUCE_LOW_MID_TX_POWER   := 1
# Tx Alarm 600 B
ENABLE_ALARM                     := 0
ENABLE_TX1750                    := 0
# MDC1200 2.8 kB
ENABLE_MDC1200                   := 1
ENABLE_MDC1200_SHOW_OP_ARG       := 1
ENABLE_PWRON_PASSWORD            := 0
ENABLE_RESET_AES_KEY             := 0
ENABLE_BIG_FREQ                  := 0
# smaa bolf 580 B
ENABLE_SMALL_BOLD                := 0
# smallest font 2 kB
ENABLE_SMALLEST_FONT             := 0
# trim trailing 44 B
ENABLE_TRIM_TRAILING_ZEROS       := 0
ENABLE_KEEP_MEM_NAME             := 1
ENABLE_WIDE_RX                   := 1
ENABLE_TX_WHEN_AM                := 0
# Freq calibration 188 B
ENABLE_F_CAL_MENU                := 0
ENABLE_TX_UNLOCK                 := 1
ENABLE_CTCSS_TAIL_PHASE_SHIFT    := 1
ENABLE_CONTRAST                  := 1
ENABLE_BOOT_BEEPS                := 0
ENABLE_DTMF_CALL_FLASH_LIGHT     := 1
ENABLE_FLASH_LIGHT_SOS_TONE      := 1
ENABLE_SHOW_CHARGE_LEVEL         := 1
ENABLE_REVERSE_BAT_SYMBOL        := 0
ENABLE_FREQ_SEARCH_TIMEOUT       := 0
ENABLE_CODE_SEARCH_TIMEOUT       := 0
ENABLE_SCAN_IGNORE_LIST          := 1
# Kill and Revive 400 B
ENABLE_KILL_REVIVE               := 0
# AM Fix 800 B
ENABLE_AM_FIX                    := 1
ENABLE_AM_FIX_SHOW_DATA          := 0
# Squelch 12 B .. can't be right ?
ENABLE_SQUELCH_MORE_SENSITIVE    := 1
ENABLE_SQ_OPEN_WITH_UP_DN_BUTTS  := 0
ENABLE_FASTER_CHANNEL_SCAN       := 1
ENABLE_COPY_CHAN_TO_VFO_TO_CHAN  := 1
# Rx Signal Bar 400 B
ENABLE_RX_SIGNAL_BAR             := 1
# Tx Audio Bar 300 B
ENABLE_TX_AUDIO_BAR              := 1
# Side Button Menu 300 B
ENABLE_SIDE_BUTT_MENU            := 1
# Key Lock 400 B
ENABLE_KEYLOCK                   := 0
ENABLE_PANADAPTER                := 0
#ENABLE_SINGLE_VFO_CHAN          := 0

OneOfEleven, 2023-11-05:
fixed.

Thank you !

closed 2023-11-05 by OneOfEleven

mjwgeek, 2023-11-05:
You're welcome!