OneOfEleven/uv-k5-firmware-custom #306: General build issues on windows

wmarkow, 2023-11-06:
I wanted to build the project on windows machine and faced a few issues:

 1. when executing win_make.bat I got the issue related to the missing **libintl3.dll** and **libiconv2.dll** libraries
 2. after fixing the issue with missing libraries I got further weird issues like:
 ```
process_begin: CreateProcess(NULL, rm -f firmware.bin firmware.packed.bin firmware start.o init.o, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [clean] Błąd 2
```
or
```
process_begin: CreateProcess(NULL, arm-none-eabi-gcc -c -mcpu=cortex-m0 start.S -o start.o, ...) failed.
make (e=2): The system cannot find the file specified.
```
or
```
c:\prj\dev-test\sources\various\uv-k5-firmware-custom>arm-none-eabi-gcc -c -mcpu=cortex-m0 start.S -o start.o
arm-none-eabi-gcc: fatal error: cannot execute 'cc1': CreateProcess: No such file or directory
compilation terminated.
```

The first issue I fixed with downloading the **Dependency.zip** libraries of **make** from https://gnuwin32.sourceforge.net/packages/make.htm web page, extracting it somewhere and adding its **bin** directory to the classpath. I didn't install the **gnu_make-3.81.exe** as its written in the main README.md; I just downloaded the zipped file and extracted it somewhere. 

The second issue in my opinion is because of the quotation marks in the **win_make.bat** around the **PATH** variable. I removed them and the compilation works perfect now:
```
@set PATH=c:\prj\dev-test\tools\gcc-arm-none-eabi-10.3-2021.10\bin\;%PATH%
@set PATH=c:\prj\dev-test\tools\gcc-arm-none-eabi-10.3-2021.10\arm-none-eabi\bin\;%PATH%
@set PATH=c:\prj\dev-test\tools\make-3.81-bin\bin\;%PATH%
@set PATH=c:\prj\dev-test\tools\make-3.81-dep\bin\;%PATH%
```

OneOfEleven, 2023-11-06:
oops

I've updated win_make.bat as per your message.

OneOfEleven, 2023-11-06:
I put the quotation marks around the paths because often windows directories have spaces in them, mine do, and bat files and command lines don't like directories with spaces without quotation marks.

closed 2023-11-06 by OneOfEleven

wmarkow, 2023-11-06:
Thanks a lot for clarification. Indeed, since long time I avoid to put spaces in the directories paths, especially on windows (but also on linux).

Let's keep this issue as a quick help for a beginners in this project (like me :)), so they don't spend lots of time on finding out why the compilation with **win_make.bat** doesn't work.

OneOfEleven, 2023-11-06:
Not putting spaces in your directory names is no solution at all, the solution is to fix whatever can't cope with spaces (it's bad programming/coding/practices not too allow for them) because spaces in names is fully part of the OS.